TrunkRecorder/trunk-recorder #892: Not an issue: Request for info RTL-SDR configuration

bburkett52, 2024-01-02:
I have searched high and low, read all docs I can find, asked in Discord and other forums, Googled everthing and can't get an answer.

I am runnig 3- RTL-SDRv3 with trunk-recorder and am gettign great aduio, a problem that haunted me with another program.

My question is concerning the parameters that can be passed via the config.json file to the SDRs. In other software, I can set gain for mixer, LNA and VGA. In this system, it looks like I can only set GAIN, with no other options.

In the other software I was using, I had gain (mixer, LVA & VGA) settings that allowed me to pull in a more distant P25P2 site.

Is there a list of applicable keys and value ranges that can be set for RTL-SDR dongles?

Thanks for reading!

tadscottsmith, 2024-01-03:
With the osmosdr driver, only LNA gain is supported by the majority of the RTL-SDR devices. The E4000 based devices support an additional IF gain. In trunk-recorder, I believe that using either `gain` or `lnaGain` will end up seting the LNA gain. The supported values are: `0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6`, but if you enter a different number, the driver will choose the closest supported value.

On startup, in the Sources section, you should see lines similar to this that indicate the supported gain stages and values.
```
[2024-01-03 17:00:18.749546] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6
[2024-01-03 17:00:18.774472] (info)   Gain set to: 33.8
```

https://github.com/osmocom/gr-osmosdr/blob/48422c87a992a862358ef8115ccaaf93bfc1336a/lib/rtl/rtl_source_c.cc#L527-L540

bburkett52, 2024-01-10:
Thanks for the info! If I am understanding correctly, the driver only can send LNA gain settings where as the driver in other software could send other parameters such as mixer and vga?

robotastic, 2024-01-28:
yep! that is correct. With the RTL-SDR devices you can only send LNA gain - just use the `gain` setting in the config.json file to set it.

I should prob put together a table for which SDR allows what type of Gain to be set.

closed 2024-01-28 by robotastic

bburkett52, 2024-01-28:
Luke,

A table defining what options can be set for specific SDRs would be great!
I spent weeks digging, asking on forums and researching and could not find
specific gain setting(s) for RTL-SDRs. I got conflicting info from folks
trying to help.

That said, bring only able to set LNA gain has proved problematic for me in
trying to tune in a weaker P25-P2 system. Using SDRTrunk, I am able to tune
in the weaker system and successfully decode, as I can adjust, mixer, LNA
and VGA. Is there any possibility these additional adjustments could be
made in future builds of trunk-recorder?

Thanks to all of you for all your work!

Regards
~Bob

robotastic, 2024-01-28:
oh interesting! it is worth trying to set those for the rtl-sdr and see if it does anything. There is no check in place that would stop it. 
use gainSettings			{ "stageName": value}	Set the gain for any stage. The value for this setting should be passed as an object, where the key specifies the name of the gain stage and the value is the amount of gain, as an int. For example:
"gainSettings": { "IF": 10, "BB": 11},

report back if it does anything

bburkett52, 2024-01-28:
Luke,

I will try that tomorrow and report back.

Am I correct that trunk-recorder uses the OSMOSDR driver and SDRtrunk uses
LibUsb for RTL-SDR devices? Could this be the difference? Obviously I am
not an expert at this stuff and I greatly appreciate the help!!

Regards,
Bob

bburkett52, 2024-01-29:
Luke,

I tried the following config.json settings for Mixer, LNA and VGA which work in SDRTrunk:
"gainSettings": { "MIX": 16, "LNA": 334, "VGA": 154},

I did not receive any errors, but it showed these results:
[2024-01-29 16:16:20.460507] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6 
[2024-01-29 16:16:20.487619] (info)   LNA Gain set to: 49.6
[2024-01-29 16:16:20.514908] (info)   MIX Gain set to: 15.7
[2024-01-29 16:16:20.541953] (info)   VGA Gain set to: 49.6
[2024-01-29 16:16:20.541997] (info)   Auto gain control is OFF

I'm not sure what to think.

Regards,
Bob