TrunkRecorder/trunk-recorder #894: 5.0-max crash with decodeTPS: true

ZeroChaos-, 2024-01-04:
```
thread 1 "trunk-recorder" received signal SIGABRT, Aborted.
0x00007ffff5eafaec in ?? () from /lib64/libc.so.6
(gdb) bt
#0  0x00007ffff5eafaec in ?? () from /lib64/libc.so.6
#1  0x00007ffff5e61246 in raise () from /lib64/libc.so.6
#2  0x00007ffff5e4a877 in abort () from /lib64/libc.so.6
#3  0x00007ffff60f89bf in std::__glibcxx_assert_fail (file=<optimized out>, line=<optimized out>, function=<optimized out>, condition=<optimized out>)
    at /var/tmp/portage/sys-devel/gcc-13.2.1_p20230826/work/gcc-13-20230826/libstdc++-v3/src/c++11/debug.cc:61
#4  0x000055555566be0a in gr::blocks::tps_decoder_sink_impl::process_message(std::shared_ptr<gr::message>) ()
#5  0x000055555568032c in monitor_messages(Config&, std::shared_ptr<gr::top_block>&, std::vector<Source*, std::allocator<Source*> >&, std::vector<System*, std::allocator<System*> >&, std::vector<Call*, std::allocator<Call*> >&) [clone .constprop.0] ()
#6  0x000055555559d9c9 in main ()
```

robotastic, 2024-01-05:
hmm... I enabled tpsDecode on an Analog system and I am not able to get it to SEGFAULT, which is weird. It probably means that the decoded messages are triggering it since I am not near any channels that have this. I haven't touched any relevant code recently though.

ZeroChaos-, 2024-01-05:
:smirk: I'm not near anything that uses tps either afaict.  I just enable all the decoders on analog channels that are multiagency use in case someone does :-)

any idea how I might get you a more useful backtrace?

JoeGilkey, 2024-01-05:
I can look into it a little deeper tomorrow night. Can you record some audio from the system you were monitoring when this happened?

FDNY uses TPS if you're anywhere near NYC.

ZeroChaos-, 2024-01-05:
Caught this outside of gdb, all tps disabled, but looking real similar
```
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = unsigned char; _Alloc = std::allocator<unsigned char>; reference = unsigned char&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
zsh: IOT instruction  trunk-recorder -c moon-tcp2.json
```
Looks like possibly another https://github.com/robotastic/trunk-recorder/issues/780 bug

ZeroChaos-, 2024-01-05:
Going by assert_fail in the output, this seems likely related to https://github.com/robotastic/trunk-recorder/issues/780 and can be reproduced and found more quickly in testing with https://github.com/robotastic/trunk-recorder/pull/793

ZeroChaos-, 2024-09-08:
most assert fixes are complete at this point so I'm going to close this until I reproduce

closed 2024-09-08 by ZeroChaos-