TrunkRecorder/trunk-recorder #900: Encrypted Channels being captured and sent to Calls and RDIO

buccipgh, 2024-01-09:
I recently upgraded to 4.7.1 and now it is sending small blips of audio clips to Calls and RDIO.
I placed a minDuration of 1.5 and it seems to have stopped it from happening... But this was new...
Here is a quick log grab of an Encrypted TG. Shows it is Encrypted but it goes thru everything till it gets to the 1.5 sec timeout.
If that 1.5 min was not in there it was sending it to Calls and RDIO.

[2024-01-09 09:10:50.623926] (info)   [PA-STARNet]      47C     TG:      10246  Freq: 161.850000 MHz    Not Recording: ENCRYPTED - src: 1202779
[2024-01-09 09:10:52.016225] (info)   [PA-STARNet]      41C     TG:      10246  Freq: 161.900000 MHz    Concluding Recorded Call - Last Update: 4s      Recorder last write:20.9108     Call Elapsed: 22
[2024-01-09 09:10:52.020074] (info)   [PA-STARNet]      41C     TG:      10246  Freq: 161.900000 MHz    Call was UPDATE not GRANT
[2024-01-09 09:10:52.020392] (info)   [PA-STARNet]      41C     TG:      10246  Freq: 161.900000 MHz    Stopping P25 Recorder Num [0]   TDMA: true      Slot: 0 Hz Error: 125
[2024-01-09 09:10:52.020915] (info)   [PA-STARNet]      41C     TG:      10246  Freq: 161.900000 MHz    - Transmission src: 1202779 pos:  0.00 length:  0.20
[2024-01-09 09:10:52.021166] (error)   [PA-STARNet]     41C     TG:      10246   Freq: 161.900000 MHz    Call length: 0.2 is less than min duration: 1.5
[2024-01-09 09:11:01.776508] (info)   [PA-STARNet]      48C     TG:      10246  Freq: 161.850000 MHz    Not Recording: ENCRYPTED - src: 1202782

Its not all of them just some.
Here is a bunch that did not even try to record.

[2024-01-09 09:21:59.604941] (info)   [PA-STARNet]      119C    TG:      10246  Freq: 161.900000 MHz    Not Recording: ENCRYPTED - src: 15000080
[2024-01-09 09:22:01.093609] (info)   [PA-STARNet]      120C    TG:      10144  Freq: 161.900000 MHz    Not Recording: ENCRYPTED - src: 1200187
[2024-01-09 09:22:03.428410] (info)   [PA-STARNet]      121C    TG:      10246  Freq: 161.850000 MHz    Not Recording: ENCRYPTED - src: 1202779
[2024-01-09 09:22:10.260280] (info)   [PA-STARNet]      122C    TG:      10246  Freq: 161.850000 MHz    Not Recording: ENCRYPTED - src: 15000080
[2024-01-09 09:22:21.958442] (info)   [PA-STARNet]      123C    TG:      10101  Freq: 161.850000 MHz    Not Recording: ENCRYPTED - src: 1200736
[2024-01-09 09:22:24.410694] (info)   [PA-STARNet]      124C    TG:      10246  Freq: 161.850000 MHz    Not Recording: ENCRYPTED - src: 15000080
[2024-01-09 09:22:39.043380] (info)   [PA-STARNet]      125C    TG:      10127  Freq: 161.900000 MHz    Not Recording: ENCRYPTED - src: 1202154

Thanks.

tadscottsmith, 2024-01-09:
Short answer: if you set `newCallFromUpdate: false` in your config.json, it should behave like before.

You probably have some reception issues with your control channel. When you see `Call was UPDATE not GRANT`, it means that trunk-recorder did not capture the original call grant that was sent. In 4.7.1 calls can be started by a grant update, but the updates lack some of the important metadata about the call such as whether or not it is encrypted. It can than take a second for the voice recorder to receive and process the encryption information that is being sent on the voice channel.

buccipgh, 2024-01-09:
Thank you!  
I have just made the change and it already seems better.
Thanks again!

closed 2024-01-09 by buccipgh