TrunkRecorder/trunk-recorder #911: sigsegv in fwrite()

ZeroChaos-, 2024-01-25:
I'm not sure how much help this is, but here is a crash I caught in gdb.  I'll try to retool my system so the debugging symbols for gnuradio are available.
```
Thread 880 "transmission_93" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffd2dcd36c0 (LWP 2465)]
0x00007ffff5e999f2 in fwrite () from /lib64/libc.so.6
(gdb) bt
#0  0x00007ffff5e999f2 in fwrite () from /lib64/libc.so.6
#1  0x00005555556466f9 in gr::blocks::transmission_sink::dowork(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
#2  0x0000555555644d7f in gr::blocks::transmission_sink::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
#3  0x00007ffff6ae71cb in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) () from /usr/lib64/libgnuradio-runtime.so.3.10.9
#4  0x00007ffff6aae287 in gr::block_executor::run_one_iteration() () from /usr/lib64/libgnuradio-runtime.so.3.10.9
#5  0x00007ffff6af1365 in gr::tpb_thread_body::tpb_thread_body(std::shared_ptr<gr::block>, std::shared_ptr<boost::barrier>, int) () from /usr/lib64/libgnuradio-runtime.so.3.10.9
#6  0x00007ffff6ae855f in gr::tpb_container::operator()() () from /usr/lib64/libgnuradio-runtime.so.3.10.9
#7  0x00007ffff6ae8604 in std::_Function_handler<void (), gr::thread::thread_body_wrapper<gr::tpb_container> >::_M_invoke(std::_Any_data const&) ()
   from /usr/lib64/libgnuradio-runtime.so.3.10.9
#8  0x00007ffff6a4277c in thread_proxy () from /usr/lib64/libboost_thread.so.1.82.0
#9  0x00007ffff5eaddc3 in ?? () from /lib64/libc.so.6
#10 0x00007ffff5f2f710 in ?? () from /lib64/libc.so.6
```

robotastic, 2024-01-25:
Thanks! This definitely helps, let me go track down what my edit might have done... I think I might need to add some locks

ZeroChaos-, 2024-01-25:
I haven't crashed in 16 hours, I'm going to assume this was fixed by https://github.com/robotastic/trunk-recorder/commit/670dceea58cbcd670202c1ba502cb58de7b486cc

closed 2024-01-25 by ZeroChaos-