TrunkRecorder/trunk-recorder #913: 5.0 RC - Cannot tune to channel anymore

theficus, 2024-01-26:
Using identical configs from the latest master branch, I'm unable to lock onto any control channel. I've attached my log file for reference.

[01-25-2024_1934_00.log](https://github.com/robotastic/trunk-recorder/files/14060035/01-25-2024_1934_00.log)

Previously discussed here: https://github.com/robotastic/trunk-recorder/issues/836#issuecomment-1882370439 but I gave it another attempt with the latest RC branch since the test branch I had been using seems to be gone now.

robotastic, 2024-01-28:
Can you include a copy of your config file? I will give a try running it with a local system and see what happens.

robotastic, 2024-01-29:
Just patched up a bug that might have been causing this. Can you try pulling the latest in this branch and giving it a try?

theficus, 2024-01-30:
I'll give it a try and report back. Thanks!

theficus, 2024-01-30:
@robotastic works perfectly after your latest commits. Thanks!

closed 2024-01-30 by theficus