TrunkRecorder/trunk-recorder #914: V4.7.1 - p25 clicking on UDP stream VS wav that is good

toddmay222, 2024-01-26:
./recorder -v
Trunk-Recorder: 4.7.1
        commit 648524ddefc336f66ec2ec3832a93d33219a62bc (master)
        describe LTS-265-g648524d
        Author: Anthony Kava <github@tonykava.com>
        Date: 2024-01-19 17:35:03 -0500

        Fix minor typos in debug logging for opcode 0x28 (#905)

**CONCERN/ISSUE:**
Recently upgraded trunk-recorder and utilized stream via UDP to a PTT app.

Clicking is evident on the audio being sent over UDP, but NOT found within the WAV file or m4a files.

To re-create:

1 - Create standard P25 config.

2 - Establish a UDP stream output:
    "plugins": [
     {
          "name":"simplestream",
          "library":"libsimplestream.so",
          "streams":[{
            "TGID":0,
            "address":"127.0.0.1",
            "port":9123,
            "sendJSON":true,
            "sendTGID":true}]
     }
    ],

3 - Alter or remove your digitalLevels (makes no difference.  1-16 or completely removed)

4 - Sniff the wire using wiresharck or TCPdump to capture the UDP packets being sent out of Trunk-Recorder to whatever local app is listening.

5 - Make sure audioArchive is true so you can also get the output to files.

Compare the two audio sources.  The outcome should appear to be:
WAV = GOOD
m4a = GOOD
UDP Output = CLICKING IN AUDIO

ASSUMPTIONS
I would ASSUME that the output WAV audio is being filters, cleaned, or processed through some functions or routines that the UDP audio is not.

It would be really nice if the UDP streaming plugin worked without the clicks... Any tips or wisdom is welcomed. :)

kylegibbons, 2024-02-15:
I am having a similar problem with analog audio over UDP.  I am sending the bits directly to a file and them playing them back with this command ```ffplay -f s16le -ar 16000 -ac 1 -i .\test_with_clicks.pcm```

I hear clicks/pops/hiss throughout the length of the file, I have attached it here so you can hear it.
[test_with_clicks.zip](https://github.com/robotastic/trunk-recorder/files/14302070/test_with_clicks.zip)

Like OP, the files produced by trunk-recorder do not have these issues.

robotastic, 2024-12-14:
@aaknitt do you have any thoughts on this? I am going to close for now since it is plug-in related, but please re-open if you have a solution. The Stream Plugin hook is right before the file writing section so that is not much more that can be done on the TR side.

closed 2024-12-14 by robotastic

aaknitt, 2024-12-14:
The clicks and pops being heard are very likely the extra metadata that's being sent along with the audio.  Programs like ffmpeg or pulseaudio are expecting nothing but audio and will interpret the extra metadata as audio, resulting in some garbage with each audio frame.  The metadata is useful for custom scripts or programs that can extract it and separate it from the audio, but if you're streaming to a program that's only expecting audio and nothing else, you should set both `sendJSON` and `sendTGID` to `false` in your config file to eliminate the metadata.  I think that should clear this up for you.  @toddmay222 @kylegibbons