TrunkRecorder/trunk-recorder #917: New Feed getting API File Upload Fail [500] Error

mfd229, 2024-02-06:
I just setup a feed and am trying to get it fed through SDR Trunk on a Windows 10 computer. I have a successful feed uploading to Broadcastify using the same application, however I get this error message when trying to upload to OpenMHz:

"ERROR i.g.d.a.b.o.OpenMHzBroadcaster - OpenMHZ API file upload fail [500] response [Talkgroup does not exist, skipping.]

I have uploaded a talkgroups list and have quadruple checked the configuration settings. Any idea what causes this error?

tadscottsmith, 2024-02-06:
This probably belongs in the SDRTrunk or trunk-server repository, but I can try to help. Can you update your system on OpenMHz and temporarily make sure `Ignore Unknown Talkgroups` is not checked? That should let you see the exact format that OpenMHz is receiving the talkgroup in.

mfd229, 2024-02-06:
It looks like it was the ignore unknown talkgroups checkbox. It is uploading now. What does that box do?

tadscottsmith, 2024-02-06:
It allows OpenMHz to accept the call even if the talkgroup from the call doesn't match one that you've defined. What type of system are you recording? And would you be willing to post what system you're uploading to on OpenMHz?

mfd229, 2024-02-06:
[Wash Co ARMER.csv](https://github.com/robotastic/trunk-recorder/files/14173797/Wash.Co.ARMER.csv)

Here is the file I am using to upload with. Here is the new system I added: https://openmhz.com/system/mnwashco

tadscottsmith, 2024-02-06:
There's a lot of activity on the Washington County towers that is above and beyond the talkgroups that you uploaded in the CSV. If you only care about the talkgroups in the CSV you can turn `Ignore Unknown Talkgroups` back on and I believe it should work, but you'll see an error on the SDRTrunk side for any talkgroup that doesn't exist in your system on the OpenMHz side.

mfd229, 2024-02-06:
I've been playing around with that checkbox. When I check it, I only get upload errors and absolutely nothing uploads, even the talkgroups I have in the list.

tadscottsmith, 2024-02-06:
Okay - that's probably a bug I missed on the SDRTrunk side. In SDRTrunk you can choose what what aliases get sent to OpenMHz. Do you have only the aliases listed in your CSV set to stream to OpenMHz?

mfd229, 2024-02-06:
No, I have everything being sent currently. I guess the override would be to manually select the alias in SDRTrunk vs uploading the file on OpenMHz

tadscottsmith, 2024-02-06:
Yep, I think you currently have to chose from one of these two options:

1. Check `Ignore Unknown Talkgroups` in OpenMHz and continue streaming everything from SDRTrunk. This will result in all calls that SDRTrunk receives showing up in OpenMHz and many of them will not have Alpha Tags.
2. Uncheck `Ignore Unknown Talkgroups` in OpenMHz and on the SDRTrunk side, only stream aliases that you've defined in OpenMHz. This should allow you to upload calls from relevant talkgroups without getting errors and the other traffic on the towers will be ignored.

mfd229, 2024-02-06:
Thanks for working through this with me! I appreciate it.

tadscottsmith, 2024-09-24:
@mfd229 would you be willing to close this issue?

closed 2024-12-14 by robotastic