TrunkRecorder/trunk-recorder #920: SampleRate wont go any higher then 10mhz

s3m1s0n1c, 2024-02-12:
Hey Brains Trust,

I am using RC5.0 branch of TR

Having an issue with HackRF where if i set the rate as ```"rate": 10000000,``` is wont decode anything

if i set the rate to  ```"rate": 9000000,``` then it will decode

a rate of  ```"rate": 10000000,``` seems to work for AirspyR2 tho

Unfortunately i need 12mhz to cover all the sites

Thanks Sonic

s3m1s0n1c, 2024-02-12:
Here is the config I am using

```
{
    "ver": 2,

    "sources": [
{
       "center": 419668750,
        "rate": 12000000,
        "error": 0,
        "ppm": 0,
        "gain": 62,
        "rfGain": 14,
        "ifGain": 62,
        "bbGain": 62,
        "digitalRecorders": 5,
        "driver": "osmosdr",
        "device": "hackrf"
    }
],
    "systems": [
{

        "control_channels": [423600000, 423350000],
        "type": "p25",
        "squelch": -100,
        "audioArchive": false,
        "callLog": false,
        "transmissionArchive": false,
        "minDuration": 0,
        "recordUnknown": true,
        "hideEncrypted": true,
        "talkgroupsFile": "scanlist.csv",
        "shortName": "sonic",
        "apiKey": "Insert Api Key Here"
    }

],

        "captureDir": "/audio/",
        "logFile": true,
        "defaultMode": "digital",
        "frequencyFormat": "mhz",
        "uploadServer": "https://api.openmhz.com"
}
```

robotastic, 2024-02-12:
@s3m1s0n1c Can you try removing the "squelch"? Since this is a trunk system it shouldn't be needed and that is a pretty high value

s3m1s0n1c, 2024-02-12:
@robotastic Yeah I did that and the same thing :(

s3m1s0n1c, 2024-02-13:
Done some more testing...

Tried different HackRF Firmwares on TR VM- Works on 8MHz or lower..
SDR-Trunk on OSX - Works 12MHz or lower
SDR-Trunk on Ubuntu VM same machine as TR VM - Doesn't work 8MHz or lower

Its looking towards maybe the machine :/  Nothing changed tho..
Will reboot the whole machine and see if that fixes..

USB Bandwidth issue?  I wasnt running the airspy R2 at the same time tho

s3m1s0n1c, 2024-02-13:
Issue turned out to be the host..

After rebooting the host machine the hackrf now works properly

Sorry for the hassle

closed 2024-02-13 by s3m1s0n1c