TrunkRecorder/trunk-recorder #923: add signal (and noise?) into json

ZeroChaos-, 2024-02-16:
TR now outputs signal level every three minutes in a big status report.  It would be great to add that signal information (and noise if available) into the json.  This would allow a much faster tuning as well as the ability to take a longer average across each system.

bburkett52, 2024-02-25:
I guess I am really overlooking something so where and how can I see this
signal info? Is it setup in the config.json file? Where does it show up?

Thanks for all the work on this program! I've been running the rc5 version
and updating as work progresses. I'm only monitoring 1 P25P2 system and all
is good!

Regards,
~Bob

ZeroChaos-, 2024-09-08:
complete, in 5.0 release

closed 2024-09-08 by ZeroChaos-