TrunkRecorder/trunk-recorder #928: streaming `P25 Parse error` with no p25 channels enabled

ZeroChaos-, 2024-02-21:
Trying to redo my config files to be a bit more modern and finally switch over to using the channelsFile to name channels and tag them correctly, and I get this:
```
[2024-02-20 19:18:53.300586] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:53.300616] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:53.926914] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:54.300854] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:54.341284] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:54.947787] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:55.331348] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:55.381811] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:55.957563] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:56.341270] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:56.421923] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:56.967843] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:57.352169] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:57.453210] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:57.967904] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:58.352148] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:58.463109] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:59.008724] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:59.392762] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:18:59.463424] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:00.039551] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:00.422961] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:00.473526] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:01.049459] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:01.433298] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:01.503893] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:02.059677] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:02.433667] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:02.504217] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:03.059771] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:03.443837] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:03.514618] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:04.099719] (error)   P25 Parse error, s:  Len: 0
[2024-02-20 19:19:04.473316] (error)   P25 Parse error, s:  Len: 0
```
This seems odd to me, because I have no p25 anything enabled
```
{
    "ver": 2,
    "sources": [
        {
            "center": 494931250.0,
            "rate": 1416000,
            "gain": 15,
            "driver": "osmosdr",
            "device": "rtl=7,buflen=32768,buffers=32"
        }
    ],
    "systems": [
      {
        "shortName": "aces1",
        "type": "conventional",
        "channelFile": "aces1.csv",
        "squelch": -35,
        "uploadScript": "../encode-upload.sh",
        "compressWav": false,
        "unitTagsFile": "../units/default.csv",
        "minDuration": 1,
        "talkgroupDisplayFormat": "tag_id",
        "decodeMDC": true,
        "decodeFSync": true,
        "decodeStar": true,
        "decodeTPS": true
      },
      {
        "shortName": "aces2",
        "type": "conventional",
        "channelFile": "aces2.csv",
        "squelch": -35,
        "uploadScript": "../encode-upload.sh",
        "compressWav": false,
        "unitTagsFile": "../units/default.csv",
        "minDuration": 1,
        "talkgroupDisplayFormat": "tag_id",
        "decodeMDC": true,
        "decodeFSync": true,
        "decodeStar": true,
        "decodeTPS": true
      },
      {
        "shortName": "npd11",
        "type": "conventional",
        "channelFile": "npd11.csv",
        "squelch": -35,
        "uploadScript": "../encode-upload.sh",
        "compressWav": false,
        "unitTagsFile": "../units/default.csv",
        "minDuration": 1,
        "talkgroupDisplayFormat": "tag_id",
        "decodeMDC": true,
        "decodeFSync": true,
        "decodeStar": true,
        "decodeTPS": true
      },
      {
        "shortName": "acpd",
        "type": "conventional",
        "channelFile": "acpd.csv",
        "squelch": -35,
        "uploadScript": "../encode-upload.sh",
        "compressWav": false,
        "unitTagsFile": "../units/default.csv",
        "minDuration": 1,
        "talkgroupDisplayFormat": "tag_id",
        "decodeMDC": true,
        "decodeFSync": true,
        "decodeStar": true,
        "decodeTPS": true
      },
      {
        "shortName": "nfd",
        "type": "conventional",
        "channelFile": "nfd.csv",
        "squelch": -35,
        "uploadScript": "../encode-upload.sh",
        "compressWav": false,
        "unitTagsFile": "../units/default.csv",
        "minDuration": 1,
        "talkgroupDisplayFormat": "tag_id",
        "decodeMDC": true,
        "decodeFSync": true,
        "decodeStar": true,
        "decodeTPS": true
      },
      {
        "shortName": "nfop5",
        "type": "conventional",
        "channelFile": "nfop5.csv",
        "squelch": -35,
        "uploadScript": "../encode-upload.sh",
        "compressWav": false,
        "unitTagsFile": "../units/default.csv",
        "minDuration": 1,
        "talkgroupDisplayFormat": "tag_id",
        "decodeMDC": true,
        "decodeFSync": true,
        "decodeStar": true,
        "decodeTPS": true
      },
      {
        "shortName": "qvps",
        "type": "conventional",
        "channelFile": "qvps.csv",
        "squelch": -35,
        "uploadScript": "../encode-upload.sh",
        "compressWav": false,
        "unitTagsFile": "../units/default.csv",
        "minDuration": 1,
        "talkgroupDisplayFormat": "tag_id",
        "decodeMDC": true,
        "decodeFSync": true,
        "decodeStar": true,
        "decodeTPS": true
      }
    ],
  "captureDir": "/tmp/tr/494",
  "callTimeout": 2,
  "frequencyFormat": "mhz"
}
```

ZeroChaos-, 2024-02-21:
Discord is blaming "decodeTPS": true
I still feel like that's a bug though.

ZeroChaos-, 2024-09-08:
The new code provides more useful output
```
[2024-09-08 11:30:39.873229] (error)   TPS Decode error, Message:  Len: 0 is < 2
[2024-09-08 11:30:40.880812] (error)   TPS Decode error, Message:  Len: 0 is < 2
[2024-09-08 11:30:41.887714] (error)   TPS Decode error, Message:  Len: 0 is < 2
[2024-09-08 11:30:42.895081] (error)   TPS Decode error, Message:  Len: 0 is < 2
[2024-09-08 11:30:43.902111] (error)   TPS Decode error, Message:  Len: 0 is < 2
[2024-09-08 11:30:44.909372] (error)   TPS Decode error, Message:  Len: 0 is < 2
[2024-09-08 11:30:45.916587] (error)   TPS Decode error, Message:  Len: 0 is < 2
[2024-09-08 11:30:46.924151] (error)   TPS Decode error, Message:  Len: 0 is < 2
[2024-09-08 11:30:47.921210] (error)   TPS Decode error, Message:  Len: 0 is < 2
[2024-09-08 11:30:48.938793] (error)   TPS Decode error, Message:  Len: 0 is < 2
[2024-09-08 11:30:49.936184] (error)   TPS Decode error, Message:  Len: 0 is < 2
```
I think it's safe to suppress this message when length is 0, no?

Dewey3, 2024-09-18:
I am getting a zero length P25 parse error on a regular basis from the latest build:
```
[2024-09-18 16:36:05.337451] (error)   P25 Parse error, s:  Len: 0
[2024-09-18 16:36:07.008117] (error)   P25 Parse error, s:  Len: 0
[2024-09-18 16:37:21.608666] (error)   P25 Parse error, s:  Len: 0
[2024-09-18 16:37:22.635738] (error)   P25 Parse error, s:  Len: 0
[2024-09-18 16:37:32.784944] (error)   P25 Parse error, s:  Len: 0
[2024-09-18 16:37:41.454473] (error)   P25 Parse error, s:  Len: 0
[2024-09-18 16:37:55.337422] (error)   P25 Parse error, s:  Len: 0
[2024-09-18 16:39:05.308059] (error)   P25 Parse error, s:  Len: 0
```

robotastic, 2024-12-14:
made it so it only prints when the len is greater that 0...

closed 2024-12-14 by robotastic