TrunkRecorder/trunk-recorder #929: new signal and noise measures have excessive precision

ZeroChaos-, 2024-02-21:
```
[2024-02-20 20:44:40.511970] (info)   [qvps]    5C      TG:  Quaker Valley Pub Sfty (         1)        Freq: 494.412500 MHz    Concluding Recorded Call - Last Update: 4354s   Recorder last write:1.70848e+09       Call Elapsed: 4354       Signal: 9.49527e-40dBm  Noise: 1.62539e-38dBm
```
I'm not great at scientific notation, but something looks really excessive here with last write, signal, and noise each being like 12 characters long.  Maybe truncate to ints?  or like 2 decimal places max

robotastic, 2024-02-21:
This should be fixed in the latest version - it may have not finished building the container yet when you tried. I add a floor() to that statement so it should just give you an int. Those values seem sort of weird though? I am bad at scientific notation though

ZeroChaos-, 2024-02-21:
I don't use the docker container, but I haven't built the last few commits.  I will update and give it a go.

ZeroChaos-, 2024-02-21:
A few more places seem to have the issue still (not always, but sometimes)
```
[2024-02-21 09:57:18.281195] (info)     [ 12 ] AnalogC  Enabled - Freq: 460.325000 MHz   Detected Signal: -42dBM (Threshold: 9.59317e+08dBM)
[2024-02-21 09:58:46.341459] (info)     [ 18 ] AnalogC  Enabled - Freq: 460.025000 MHz   Detected Signal: -34dBM (Threshold: -42dBM)
```

ZeroChaos-, 2024-03-28:
this has been fixed a while now

closed 2024-03-28 by ZeroChaos-