TrunkRecorder/trunk-recorder #932: simplestream plugin improvements

aaknitt, 2024-03-02:
Adding for comments and tracking, I'll work on this one.  

From Discord:

> bobert5064 — Yesterday at 5:13 PM
> I'm no c++ expert, and it may not be related to the issue I'm experiencing, but could someone explain why the foreach looks are using pointers to stream and TGID when they aren't being modified in the loops? I'm wondering if potentially the value of the pointer is somehow getting changed before the stream it written to the socket in cases where TR is especially busy and audio_stream is being called often. Also when looking at the code that was added for JSON support I see that TGID is passed to the JSON by value, but to the stream buffer by reference. The json_string is also pushed to the buffer by value, but curiously the length of the JSON is passed by ref. I might just be out of my depths here and it all makes perfect sense, but either way I'd like to learn why.

> lilhoser — Yesterday at 6:20 PM
> I just wrote a plugin called callstream you might want to look at. I considered using SS first but it didn’t quite fit my needs. I’m not 100% sure how parallelized TR plugin callbacks are but it seems to me that a plugins can be invoked by multiple TR threads processing sample data from different calls in progress. And SS is not thread safe - inside its audio_stream callback it calls push_back into a vector without a lock. Unless I’m misunderstanding, this can result in unpredictable behavior. I added locking to my plugin

aaknitt, 2024-03-02:
Working branch for this issue:
https://github.com/aaknitt/trunk-recorder/tree/simplestream-fixes

aaknitt, 2024-03-02:
Making a note to add this FFMPEG example to the documentation with this update:

bobert5064 — Yesterday at 3:18 PM
Hi @lilhoser , here's an FFMPEG command that takes PCM audio from SS on stdin, cleans it up, and outputs ogg/opus to stdout. Your usecase probably differs slightly, but this should point you in the right direction: ffmpeg -loglevel warning -f s16le -ar 16000 -ac 1 -i pipe:0 -af:a adeclick -f:a ogg -c:a libopus -frame_duration:a 20 -vbr:a on -b:a 48000 -application:a voip pipe:1

aaknitt, 2024-06-28:
Also want to add additional JSON metadata on call_start (talkgroup names, etc.) and provide a JSON package on call_end

robotastic, 2024-12-14:
Did this get addressed? I seem to remember a PR for simplestream

aaknitt, 2024-12-14:
Yes, these have all been done, closing.

closed 2024-12-14 by aaknitt