TrunkRecorder/trunk-recorder #934: Segmentation fault on rates higher than 10 MHz

srs4511351, 2024-03-10:
I have a Raspberry Pi 4 with Debian Bookworm.
I have trunk-recorder working on my HackRF up to 10 MHz sample rate, but if I set it higher, I get the Segmentation fault and these errors in the terminal.
```
...
[2024-03-10 01:06:50.924596] (info)   [MPSCS]	Started with Control Channel: 853.250000 MHz
[2024-03-10 01:06:50.925459] (info)   	 P25 Trunking two-stage decimator - Initial decimated rate: 96000 Second decimated rate: 24000 FA: 6250 FB: 12000 System Rate: 12000000
[2024-03-10 01:06:50.943820] (info)   	 P25 Trunking ARB - Initial Rate: 12000000 Resampled Rate: 24000 Initial Decimation: 125 System Rate: 24000 ARB Rate: 1
Segmentation fault
```
Here is my config file:
```
{
    "ver": 2,

    "sources":   [{
        "center": 856150000,
        "rate": 12000000,
        "error": 300,
        "ppm": 0,
        "gain": 0,
        "rfGain": 0,
	"ifGain": 40,
	"bbGain": 18,
        "digitalRecorders": 2,
        "driver": "osmosdr",
	"device": "hackrf"
    }],
    "systems": [{
        "control_channels": [853250000, 855087500, 855287500],
        "type": "p25",
	"talkgroupsFile": "michigan_talkgroups.csv",
	"recordUnknown": true,
        "shortName": "MPSCS",
	"hideUnknownTalkgroups": false,
        "modulation": "qpsk",
        "audioArchive": false,
        "transmissionArchive": false
    }],
    "plugins": [
	{
          "name": "rdioscanner_uploader",
          "library": "librdioscanner_uploader.so",
          "server": "http://127.0.0.1:3000",
          "systems": [{
                  "shortName": "MPSCS",
                  "apiKey": "9e5cf154-7456-42d8-8bb5-8ba38a2c3034",
                  "systemId": 796
          }]
	}
    ],
"broadcastSignals": false
}
```
In issue  #920, it looks like someone was able to do it.

Does anyone know why this is happening?

robotastic, 2024-03-10:
What version of Trunk Recorder are you using? Is it the Master branch off of GitHub? Can you try capturing the Core dump and running GDB to figure out where it is crashing?

srs4511351, 2024-03-11:
Trunk-Recorder: Version: 4.7.1
The branch is 'origin/master'

On my system `ulimit -c` prints 0, which is the maximum size of core files created.

I have attached the output from gdb.
Please let me know if this is good enough.

[gdb.txt](https://github.com/robotastic/trunk-recorder/files/14553441/gdb.txt)

srs4511351, 2024-03-11:
The above process ended in:
```
Thread 18 "hackrf_source_c" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fdfffec40 (LWP 2491)]
0x0000007ff6dd1f78 in hackrf_source_c::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) ()
   from /usr/local/lib/aarch64-linux-gnu/libgnuradio-osmosdr.so.0.2.0
```
To get the debug info, I typed 
`thread apply all bt`

adeweever91, 2024-05-15:
were you able to resolve this? I have a similar issue when running in docker. i saw some posts that seemed to indicate it can help to change the hackrf firmware but havent tried any different versions yet

srs4511351, 2024-06-12:
I have not been able to get HackRF working on sample rates higher than 10 MHz with trunk-recorder.

I do keep my firmware up to date, but it works with SDRglut, gqrx and other SDR software at the full 20 MHz, so that isolates the problem to trunk-recorder.

srs4511351, 2024-06-12:
Have you tried 20 MHz with the soapy device?
I haven't tried that yet.

closed 2025-01-15 by robotastic

V3ntus, 2025-07-24:
Having this issue on my Mac M1

Trunk-Recorder: 5.0.0
MacOS 15.5
libhackrf version: 2024.02.1 (0.9)
HackRF Firmware Version: 2024.02.1 (API:1.08)
HackRF Hardware Revision: r10

```
{
    "ver": 2,
    "sources": [
        {
            "center": 856434375.0,
            "rate": 12000000,
            "ppm": -20,
            "gain": 40,
            "agc": false,
            "antenna": "TX/RX",
            "digitalRecorders": 12,
            "driver": "osmosdr",
            "device": "hackrf=4e3463"
        }
    ],
    "systems": [
        {
            "control_channels": [
                852725000,
                853137500,
                853312500,
                853575000
            ],
            "type": "p25",
            "digitalLevels": 1,
            "modulation": "fsk4",
            "hideEncrypted": true,
            "talkgroupDisplayFormat": "id_tag",
            "compressWav": true
        }
    ],
    "logLevel": "trace",
    "frequencyFormat": "mhz"
}
```