TrunkRecorder/trunk-recorder #938: Installation Error on Latest Version of RaspberryPiOS

Dewey3, 2024-03-20:
While doing an `update` and `upgrade` to one of my 8GB RPi 4's last week, the attempted upgrade kept erroring out.  Doing a little GoogleFoo, I saw somewhere where the RPi OS was breaking when upgrading (the new kernal is `Linux kernel 6.6.20 - 6f16847710cc0502450788b9f12f0a14d3429668`).  So this week, I downloaded the latest RaspberryPiOS, dated 3/15/2024, and started fresh.  Now, when going through the "Add all of the libraries needed" portion of the TR install, it keeps ending with:
```
Errors were encountered while processing:
 xtrx-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

I went back through the installation and found were the error is occurring.  This is what I am seeing:
```
Building for 6.6.20+rpt-rpi-v8
Building initial module for 6.6.20+rpt-rpi-v8
Error! Bad return status for module build on kernel: 6.6.20+rpt-rpi-v8 (aarch64)
Consult /var/lib/dkms/xtrx/0.0.1+git20190320.5ae3a3e-3.2/build/make.log for more information.
dpkg: error processing package xtrx-dkms (--configure):
 installed xtrx-dkms package post-installation script subprocess returned error exit status 10
```

I looked for the `make.log` referenced in the error, but the `/var/lib/dkms...` directory stops after the `...-3.2`.  There is no `/build/` directory nor `make.log` there.  Does anyone know if this is going to be a problem.  While I have not yet run TR, the "Build" process did complete without any errors.

Dewey3, 2024-03-20:
Update:  I rebooted and just performed another `update` and `upgrade`.  I am now getting errors similar to the errors that I got when I first tried to update before grabbing the latest RaspberryPiOS.  I already knew that the errors were a RPiOS induced error during the TR installation and not TR itself, but just wanted to know if it would cause a TR problem.  If anyone is interested, this is the error:
```
console@TravelPi:~ $ sudo apt update
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:5 https://www.deb-multimedia.org bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
console@TravelPi:~ $ sudo apt full-upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up xtrx-dkms (0.0.1+git20190320.5ae3a3e-3.2) ...
Removing old xtrx-0.0.1+git20190320.5ae3a3e-3.2 DKMS files...
Deleting module xtrx-0.0.1+git20190320.5ae3a3e-3.2 completely from the DKMS tree.
Loading new xtrx-0.0.1+git20190320.5ae3a3e-3.2 DKMS files...
Building for 6.6.20+rpt-rpi-v8
Building initial module for 6.6.20+rpt-rpi-v8
Error! Bad return status for module build on kernel: 6.6.20+rpt-rpi-v8 (aarch64)
Consult /var/lib/dkms/xtrx/0.0.1+git20190320.5ae3a3e-3.2/build/make.log for more information.
dpkg: error processing package xtrx-dkms (--configure):
 installed xtrx-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 xtrx-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
console@TravelPi:~ $
```

Dewey3, 2024-03-20:
Hopefully the last update:  after more GoogleFoo, I got the idea to uninstall `dkms` since I was running some tests and had nothing to lose.  So I completely removed and purged `dkms` then did another `update` and `upgrade` and lo and behold, no update errors.  I then reinstalled `dkms` and did a final `update` and `upgrade` and everything appears, and hopefully stays well.  If time allows, I will start the TR installation over tomorrow from a fresh RPiOS, but doing the uninstall then reinstall of `dkms` and will see how that works.

Dewey3, 2024-03-22:
Ok, here is what I've come up with.  Being an intermediate Linux user, my way of trying to troubleshoot and solve this was through process of elimination.  First, I did discover that something in the "Add all of the libraries needed" portion is what causes `dkms` and `xtrx-dkms` to be installed.  **The problem is that something is broken in the way `xtrx-dkms` works with the latest release of RPiOS** resulting in constant errors when installing packages.

So on a brand new RPiOS installation (each time I would perform an elimination), I continually checked for the presence of `dkms` and `xtrx-dkms` which did not show up until after completing the "Add all of the libraries needed" portion of the installation.  I first tried uninstalling `xtrx-dkms` **and** `dkms` but attempting that threw errors every time I tried to reinstall `xtrx-dkms`.  So my solution was to just uninstall and reinstall `dkms` only.  After doing so, then re-running the "Add all of the libraries needed" step, I did not receive any errors for the remainder of the installation.

So with all that said, the bottom line is that it is something that breaks when `xtrx-dkms` is installed on the latest RPiOS, which in turn throws errors with most (if not all) package installations and upgrades after that.  I did find that TR worked whether I ignored the error during TR installation, or used the `dkms` uninstall-reinstall that I mentioned above.  What I like about the `dkms` uninstall-reinstall method is that I do not constantly get errors when performing an `'update` or `upgrade` or installing anything else afterwards.  I apologize for the way I had to go through this, then explain it here but as I said, I am only an intermediate Linux user.

k5map, 2024-06-06:
@Dewey3 ... what command did you use to determine if xtrx-dkms and dkms were installed?  Also, what was the uninstall command you used?

robotastic, 2024-12-14:
Is this still an issue - I think I have an up to date Pi I can test on if it is

Dewey3, 2024-12-14:
Thank you Luke! no it's no longer an issue.  Your latest revised installation instructions address the dkms issue.

closed 2024-12-14 by Dewey3