TrunkRecorder/trunk-recorder #941: Feature Request: New Directory for each TGID

decoy4fun, 2024-03-31:
This may be a niche request and not worth the time. How much work would it take to add an option in the config file to allow each talkgroup to create its own directory in your recording folder? I imagine this being set at a system level, and disabled by default unless overridden by the user. 

My use case is I like to pull audio files for events and merge them into one long audio file. Sorting/filtering isn't horrible, but if each was in its own directory it would be much easier to find the start/end and merge just those files.

ge0metrix, 2024-03-31:
I'd go as far as asking to have a format string for the audio file names ... allow them to be customized.

example
`{shortname}/{talkgroup_tag}/{timestamp}_{freq}.wav`

rosecitytransit, 2024-06-05:
One idea would be to copy just the specific talk group files to another folder or run a command on them using a wildcard (e.g. `cp shortname-1234-*`).

For my local instance (where I'm not uploading), I just edited the code to have it create what I wanted.