TrunkRecorder/trunk-recorder #946: Startup trunk-recorder and radio-scanner on RPI boot

jeffbarclay, 2024-04-10:
trying to setup a @reboot script to start trunk-recorder in one tty session and also start radio-scanner in another tty session since both need to run simultaneously. here is the script I have but seems both try to run in the initial tty window and the rdio-scanner will only run after the trunk-recorder is killed.

```
 #!/bin/bash

date >> date.txt

setsid sh -c 'exec commmand <> /dev/tty2 >&0 2>&1'
/home/pi/trunk-build/trunk-recorder --config /home/pi/trunk-build/config.json

#openvt -c 2

setsid sh -c 'exec commmand <> /dev/tty3 >&0 2>&1'
/home/pi/trunk-build/rdio-scanner/rdio-scanner
 
```

hsoj, 2024-04-10:
While I have not dug into the pipe transitions here, just based on the process order of operations, you might need to background the processes if the desire isn't to hold them open from this script.

jeffbarclay, 2024-04-10:
The problem is both scripts need to run at the same time. This configuration only fires the trunk-recorder and not the Rdio-scanner until after the trunk-recorder is killed

jeffbarclay, 2024-04-10:
tmux appears to be a way around this but I'm not sure how to script that?

hsoj, 2024-04-10:
If you're asking about a headless system on an RPI, I'm assuming it's running systemd.

Have you set these up to not run behind systemd?  How have you set these up to run initially?

penguinzephyr, 2024-04-11:
Is there a reason you arent using the example systemd service?

https://github.com/robotastic/trunk-recorder/blob/70698818065b24a494fc5488c82445609eab9ebc/examples/SETUP_SERVICE.md

You could modify this example to start rdio-scanner as a separate service too.

I personally prefer supervisor and it keeps both tr and rdio up for me happily. Example configuration.

/etc/supervisor/conf.d/trunk-recorder.conf
`[program:trunk-recorder]
user=ubuntu
redirect_stderr=true
stdout_logfile=/home/ubuntu/logs/trunk-recorder-supervisor.log
stdout_logfile_maxbytes=32MB
stdout_logfile_backups=10
directory=/home/ubuntu/
command=/bin/bash -c "/usr/local/bin/trunk-recorder -c /home/ubuntu/config.json"
numprocs=1`

/etc/supervisor/conf.d/rdio.conf
`[program:rdio]
user=ubuntu
redirect_stderr=true
stdout_logfile=/home/ubuntu/logs/rdio.log
stdout_logfile_maxbytes=32MB
stdout_logfile_backups=10
directory=/home/ubuntu/
command=/home/ubuntu/rdio-scanner
numprocs=1`

jeffbarclay, 2024-04-11:
Thanks this will work - I actually just got it working with tmux installed and scripted the tmux to startup with 2 sessions.

https://github.com/robotastic/trunk-recorder/blob/70698818065b24a494fc5488c82445609eab9ebc/examples/SETUP_SERVICE.md

Dewey3, 2024-04-11:
If it helps, just set Rdio-Scanner as a service.  That way, Rdio-Scanner will always start on reboot and your can set up a `@reboot` crontab to start TrunkRecorder on every reboot.  My `@reboot` crontab for my headless RPis point to a script containing the following text:
```
#!/bin/sh

sleep 1m

cd /home/truser/trunk-build
/home/truser/trunk-build/trunk-recorder --config={configuration file name}.json
```
-or-
You can just start your `tmux` session and manually start TrunkRecorder then "detach" since Rdio-Scanner will already be running as a service.

closed 2024-12-14 by robotastic