TrunkRecorder/trunk-recorder #952: RC5

Dewey3, 2024-04-26:
I am seeing more references to RC5 as testing continues.  Not wanting to hijack any of the other threads, I rather ask about it here.  Where can I find the latest versions of RC5?  I had a link to try a RC5 version 2 or 3 months ago, but that version never locked in on the control channels so I couldn't go any further.  I would like to give it another spin if it has evolved since then.  My apologies in advance if this information is over on Discord.  I'm kicking and screaming about going to Discord and would rather not unless it is absolutely necessary.

taclane, 2024-04-26:
It is available both as a docker image:
`docker pull robotastic/trunk-recorder:RC5.0`  etc..

Or can be compiled from source using the normal instructions:
`git clone https://github.com/robotastic/trunk-recorder.git --branch rc/v5.0 trunk-recorder-5`
(if you're unsure about breaking a working copy of trunk recorder 4.7, you may want to use a separate GitHub directory for the 5.0 rc, that way you can quickly revert by going back to your 4.7 directory and `make install` it back into place)

Dewey3, 2024-04-27:
> It is available both as a docker image: `docker pull robotastic/trunk-recorder:RC5.0` etc..
>
> Or can be compiled from source using the normal instructions: `git clone [...]

Thank you again @tadscottsmith .  I just finished the latest RC5 build and watching it decoding.  For me, the good news is that I had no problem capturing the control channel and it is running well.  The bad news is that the UIDs are still running together (not changing on quick back-to-back transmissions) on the very busy Prince George's Co, MD, 2A8 system (https://www.radioreference.com/db/sid/6341).  Also, I still get the installation errors I mentioned in https://github.com/robotastic/trunk-recorder/issues/938 when installing on a RPi.  Somewhere in the "Add all of the libraries needed" step, `xtrx-dkms` is getting installed, but `xtrx-dkms` is not working with the latest version of RaspberryPi OS (??? Debian also ???).  That causes errors to be created after anything is installed or the system is updated.  I know that it is not a Trunk-Recorder issue, but I am mentioning it since it affects the Trunk-Recorder installation.  Thanks.

closed 2024-05-03 by Dewey3