TrunkRecorder/trunk-recorder #972: SIGSEGV after v5.0.0 upgrade, master branch commit id f00a05e7cf81e9422e9cf86b63b09ec26e312170

Squirrelies, 2024-08-19:
## Issue
Experiencing intermittent crashes after upgrading to v5.0.0. DEBUG build as of commit [f00a05e7cf81e9422e9cf86b63b09ec26e312170](https://github.com/robotastic/trunk-recorder/tree/f00a05e7cf81e9422e9cf86b63b09ec26e312170)

Built using a mix of instructions from [How to Debug a Seg Fault | Trunk Recorder](https://trunkrecorder.com/docs/DEBUG) and various internet sources on applying ulimits to systemd services.

Willing to make and try new builds. Let me know if there is any additional info I can provide.

## Systemd service file
```
squirrelies@sdr1:~/trunk-recorder$ cat /etc/systemd/system/trunk-recorder.service
[Unit]
Description=Trunk-Recorder
After=network.target
Wants=network-online.target

[Service]
Type=simple
User=squirrelies
Group=squirrelies
LimitCORE=infinity
Restart=on-failure
WorkingDirectory=/home/squirrelies/tr-live
ExecStart=trunk-recorder --config=config.json

[Install]
WantedBy=multi-user.target
```

## Core dumps
```
squirrelies@sdr1:~/trunk-recorder$ ls -la /var/lib/apport/coredump/
total 12064584
drwxr-xr-x 2 root        root       4096 Aug 19 13:44 .
drwxr-xr-x 3 root        root       4096 Apr 23 09:40 ..
-r-------- 1 squirrelies root 4119076864 Aug 18 21:49 core._usr_local_bin_trunk-recorder.1000.89a599dd-081d-4a9c-bb1e-37da59950c70.155881.14653388
-r-------- 1 squirrelies root 4118155264 Aug 19 05:01 core._usr_local_bin_trunk-recorder.1000.89a599dd-081d-4a9c-bb1e-37da59950c70.293521.35614011
-r-------- 1 squirrelies root 4116881408 Aug 19 13:45 core._usr_local_bin_trunk-recorder.1000.89a599dd-081d-4a9c-bb1e-37da59950c70.346624.38200911
```

## Core dump stack trace
```
squirrelies@sdr1:~/trunk-recorder$ gdb trunk-recorder /var/lib/apport/coredump/core._usr_local_bin_trunk-recorder.1000.89a599dd-081d-4a9c-bb1e-37da59950c70.346624.38200911
GNU gdb (Ubuntu 15.0.50.20240403-0ubuntu1) 15.0.50.20240403-git
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from trunk-recorder...

warning: Can't open file /SYSV00000000 (deleted) during file-backed mapping note processing

warning: Can't open file /home/squirrelies/orcexec.B6ld7T (deleted) during file-backed mapping note processing
[New LWP 386901]
[New LWP 346624]
[New LWP 346629]
[New LWP 346630]
[New LWP 386903]
[New LWP 386647]
[New LWP 346632]
[New LWP 346628]
[... 69 lines cut ...]
Aug 19 13:44:56 sdr1 trunk-recorder[346624]: [2024-08-19 13:44:56.157811] (error)   P25 Parse error, s:  Len: 0
Aug 19 13:44:57 sdr1 trunk-recorder[346624]: [2024-08-19 13:44:57.193561] (error)   P25 Parse error, s:  Len: 0
Aug 19 13:44:58 sdr1 trunk-recorder[346624]: [2024-08-19 13:44:58.240040] (error)   P25 Parse error, s:  Len: 0
Aug 19 13:44:59 sdr1 trunk-recorder[346624]: [2024-08-19 13:44:59.012401] (error)   [wiwinnouta] Retuning to Control Channel: 858.462500 MHz
Aug 19 13:44:59 sdr1 trunk-recorder[346624]: [2024-08-19 13:44:59.012582] (info)            - System Source 0 - Min Freq: 858.075000 MHz Max Freq: 860.375000 MHz
Aug 19 13:44:59 sdr1 trunk-recorder[346624]: [2024-08-19 13:44:59.012800] (error)   [wiwinnouta]         Control Channel Message Decode Rate: 0/sec, count:  1
Aug 19 13:44:59 sdr1 trunk-recorder[346624]: [2024-08-19 13:44:59.104097] (error)   [wiwinnouta] Retuning to Control Channel: 857.012500 MHz
Aug 19 13:44:59 sdr1 trunk-recorder[346624]: [2024-08-19 13:44:59.104152] (info)            - System Source 1 - Min Freq: 855.821875 MHz Max Freq: 858.121875 MHz
Aug 19 13:44:59 sdr1 trunk-recorder[346624]: [2024-08-19 13:44:59.113530] (info)            Xlating Channelizer decimator - freq_xlating taps: 91 Decim: 4 Resampled Rate: 24000 Lowpass Taps: 83
Aug 19 13:45:05 sdr1 systemd[1]: trunk-recorder.service: Main process exited, code=dumped, status=11/SEGV
Aug 19 13:45:05 sdr1 systemd[1]: trunk-recorder.service: Failed with result 'core-dump'.
Aug 19 13:45:05 sdr1 systemd[1]: trunk-recorder.service: Consumed 8h 56min 36.054s CPU time.
Aug 19 13:45:05 sdr1 systemd[1]: trunk-recorder.service: Scheduled restart job, restart counter is at 3.
Aug 19 13:45:05 sdr1 systemd[1]: Started trunk-recorder.service - Trunk-Recorder.
Aug 19 13:45:05 sdr1 trunk-recorder[387707]: [2024-08-19 13:45:05.484693] (info)
Aug 19 13:45:05 sdr1 trunk-recorder[387707]: -------------------------------------
Aug 19 13:45:05 sdr1 trunk-recorder[387707]:      Trunk Recorder
Aug 19 13:45:05 sdr1 trunk-recorder[387707]: -------------------------------------
Aug 19 13:45:05 sdr1 trunk-recorder[387707]: [2024-08-19 13:45:05.484726] (info)
Aug 19 13:45:05 sdr1 trunk-recorder[387707]: -------------------------------------
Aug 19 13:45:05 sdr1 trunk-recorder[387707]: INSTANCE
Aug 19 13:45:05 sdr1 trunk-recorder[387707]: -------------------------------------
Aug 19 13:45:05 sdr1 trunk-recorder[387707]: [2024-08-19 13:45:05.484740] (info)   Using Config file: config.json
Aug 19 13:45:05 sdr1 trunk-recorder[387707]: [2024-08-19 13:45:05.484749] (info)   Trunk-Recorder: Version: 5.0.0
```

Squirrelies, 2024-08-20:
Details from another core dump were...

```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `trunk-recorder --config=config.json'.
Program terminated with signal SIGSEGV, Segmentation fault.
Downloading source file /usr/src/gr-osmosdr-0.2.5-2.1build3/lib/rtl/rtl_source_c.cc
#0  0x000079fab5b7cf9a in rtl_source_c::work (this=0x5eaa67fb7dc0, noutput_items=54, input_items=..., output_items=...) at /usr/src/gr-osmosdr-0.2.5-2.1build3/lib/rtl/rtl_source_c.cc:354
354           *out++ = gr_complex(_lut[buf[i * 2]], _lut[buf[i * 2 + 1]]);
[Current thread is 1 (Thread 0x79f818e006c0 (LWP 346258))]
(gdb) bt full
#0  0x000079fab5b7cf9a in rtl_source_c::work (this=0x5eaa67fb7dc0, noutput_items=54, input_items=..., output_items=std::vector of length 1, capacity 1 = {...}) at /usr/src/gr-osmosdr-0.2.5-2.1build3/lib/rtl/rtl_source_c.cc:354
        i = 0
        nout = <optimized out>
        buf = 0x79fab1b7fd3a "\200\201\202yz\200\200\202\177\177~{\202{\201}\177x\177\200}\205}\201\177x}y{|{~}yx\203\200|\204|\201\202|\203|}\204{\201~|\200\201\204\202{\177\177z\201\202\177\200\201\205}\177w{\200\206~\205~\200\200\177\202\205\202\177~}z\177~~~\203\206\202~\200}\202~}\201|\201|\202||\204\200\200\200\200{|}}\200{~|\201\204\201\205~\211\201\203\177\205\201\207\204|\177\177\202\202}u\203\177x\177}\177\200\203{\204\201\207}\202\200\177\213x\206|}\203\177\202\201~\202\177\200\203\202\200\204\202\205~\200\177\204\206\201}\201z\177\204~\202\210\200{\200\200~\201\201\204\201\177\177\203\203\177"...
        out = 0x79faa6c173b8
#1  0x000079fab556cca0 in gr::sync_block::general_work (this=0x5eaa67fb7dc0, noutput_items=<optimized out>, ninput_items=..., input_items=..., output_items=...) at /usr/src/gnuradio-3.10.9.2-1.1ubuntu2/gnuradio-runtime/lib/sync_block.cc:49
        r = <optimized out>
#2  0x000079fab553298c in gr::block_executor::run_one_iteration (this=0x79f818dff7f0) at /usr/src/gnuradio-3.10.9.2-1.1ubuntu2/gnuradio-runtime/lib/block_executor.cc:622
        i = <optimized out>
        n = <optimized out>
        msg = <optimized out>
        i = <optimized out>
        try_again = <optimized out>
        noutput_items = 54
        max_items_avail = <optimized out>
        max_noutput_items = <optimized out>
        new_alignment = 0
        alignment_state = <optimized out>
        output_idx = 0
        m = <optimized out>
        d = 0x5eaa68b26b10
#3  0x000079fab557cfa8 in gr::tpb_thread_body::tpb_thread_body (this=<optimized out>, block=..., start_sync=..., max_noutput_items=<optimized out>, this=<optimized out>, block=..., start_sync=..., max_noutput_items=<optimized out>)
    at /usr/src/gnuradio-3.10.9.2-1.1ubuntu2/gnuradio-runtime/lib/tpb_thread_body.cc:94
        s = <optimized out>
        msg = std::shared_ptr<pmt::pmt_base> (empty) = {get() = <optimized out>}
        max_nmsgs = 8192
        logger = {_name = "tpb_thread_body", d_logger = std::shared_ptr<spdlog::logger> (use count 1, weak count 0) = {get() = 0x79faa800c700}}
        d = 0x5eaa68b26b10
        p = <optimized out>
        d = <optimized out>
        s = <optimized out>
        msg = <optimized out>
        p = <optimized out>
        max_nmsgs = <optimized out>
        logger = <optimized out>
        i = <optimized out>
        __for_range = <optimized out>
        __for_begin = <optimized out>
        __for_end = <optimized out>
        guard = <optimized out>
        timeout = <optimized out>
        guard = <optimized out>
#4  0x000079fab556e93b in gr::tpb_container::operator() (this=this@entry=0x5eaa68a17040) at /usr/src/gnuradio-3.10.9.2-1.1ubuntu2/gnuradio-runtime/lib/scheduler_tpb.cc:38
        body = {d_exec = {d_block = std::shared_ptr<gr::block> (use count 9, weak count 2) = {get() = 0x5eaa67fb7dc0}, d_logger = std::shared_ptr<gr::logger> (use count 1, weak count 0) = {get() = 0x79faa8007040},
            d_debug_logger = std::shared_ptr<gr::logger> (use count 1, weak count 0) = {get() = 0x79fa9c003aa0}, d_ninput_items_required = std::vector of length 0, capacity 0, d_ninput_items = std::vector of length 0, capacity 0,
            d_input_items = std::vector of length 0, capacity 0, d_input_done = std::vector<bool> of length 0, capacity 0, d_output_items = std::vector of length 1, capacity 1 = {0x79faa6c173b0},
            d_start_nitems_read = std::vector of length 0, capacity 0, d_returned_tags = std::vector of length 0, capacity 0, d_max_noutput_items = 8192, d_use_pc = false}}
#5  0x000079fab556e9e4 in gr::thread::thread_body_wrapper<gr::tpb_container>::operator() (this=0x5eaa68a17040) at /usr/src/gnuradio-3.10.9.2-1.1ubuntu2/gnuradio-runtime/lib/../include/gnuradio/thread/thread_body_wrapper.h:49
No locals.
#6  std::__invoke_impl<void, gr::thread::thread_body_wrapper<gr::tpb_container>&> (__f=...) at /usr/include/c++/13/bits/invoke.h:61
No locals.
#7  std::__invoke_r<void, gr::thread::thread_body_wrapper<gr::tpb_container>&> (__fn=...) at /usr/include/c++/13/bits/invoke.h:111
No locals.
#8  std::_Function_handler<void (), gr::thread::thread_body_wrapper<gr::tpb_container> >::_M_invoke(std::_Any_data const&) (__functor=...) at /usr/include/c++/13/bits/std_function.h:290
No locals.
#9  0x000079fab586297f in boost::(anonymous namespace)::thread_proxy (param=<optimized out>) at libs/thread/src/pthread/thread.cpp:179
        thread_info = <optimized out>
        lock = <optimized out>
[... cut ...]
        pd = <optimized out>
        out = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {134106476185280, 3210532438489141310, 134106476185280, -216, 2, 140737219748496, 3210532438610776126, 3209732355794767934}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {
              prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
#11 0x000079fab4929c3c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
No locals.
```

robotastic, 2024-12-14:
Did you ever figure out a root cause for this? I am going to close for now to clean things up, but feel free to reopen if there are more clues

closed 2024-12-14 by robotastic