TrunkRecorder/trunk-recorder #976: 5.01 and master fail to build on latest up to date arch

mldulaney, 2024-09-27:
[ 40%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/iqfile_source_impl.cc.o
cd /home/mock/trunk-recorder/build/lib/op25_repeater/lib && /usr/bin/c++ -DBOOST_ALL_DYN_LINK -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_BIND_GLOBAL_PLACEHOLDERS -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_LOG_DYN_LINK -DBOOST_LOG_NO_LIB -DBOOST_LOG_SETUP_DYN_LINK -DBOOST_LOG_SETUP_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_RANDOM_DYN_LINK -DBOOST_RANDOM_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DDISABLE_NID_PARITY_CHECK -DEZPWD_NO_EXCEPTS -DFMT_SHARED -DGNURADIO_VERSION=200721 -DGR_CTRLPORT -DGR_MPLIB_GMP -DGR_PERFORMANCE_COUNTERS -DMANUAL_GITINFO=\"\" -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -Dgnuradio_op25_repeater_EXPORTS -I/home/mock/trunk-recorder/build -I/home/mock/trunk-recorder/build/lib -I/home/mock/trunk-recorder/build/include -I/home/mock/trunk-recorder/lib/op25_repeater/lib -I/home/mock/trunk-recorder/lib/op25_repeater/include -I/home/mock/trunk-recorder/lib -D_GLIBCXX_ASSERTIONS -pthread -Wno-narrowing -fvisibility=hidden -fPIC -DNDEBUG -std=gnu++17 -fPIC   -fvisibility=hidden -w -MD -MT lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/iqfile_source_impl.cc.o -MF CMakeFiles/gnuradio-op25_repeater.dir/iqfile_source_impl.cc.o.d -o CMakeFiles/gnuradio-op25_repeater.dir/iqfile_source_impl.cc.o -c /home/mock/trunk-recorder/lib/op25_repeater/lib/iqfile_source_impl.cc
In file included from /usr/include/fmt/format.h:41,
                 from /usr/include/fmt/core.h:5,
                 from /usr/include/spdlog/fmt/fmt.h:28,
                 from /usr/include/gnuradio/io_signature.h:19,
                 from /usr/include/gnuradio/basic_block.h:15,
                 from /usr/include/gnuradio/block.h:17,
                 from /usr/include/gnuradio/sync_block.h:15,
                 from /home/mock/trunk-recorder/lib/op25_repeater/include/op25_repeater/iqfile_source.h:27,
                 from /home/mock/trunk-recorder/lib/op25_repeater/lib/iqfile_source_impl.h:26,
                 from /home/mock/trunk-recorder/lib/op25_repeater/lib/iqfile_source_impl.cc:28:
/usr/include/fmt/base.h: In instantiation of ‘static void fmt::v11::detail::value<Context>::format_custom_arg(void*, typename Context::parse_context_type&, Context&) [with T = boost::basic_format<char>; Formatter = fmt::v11::formatter<boost::basic_format<char> >; Context = fmt::v11::context; typename Context::parse_context_type = fmt::v11::basic_format_parse_context<char>]’:
/usr/include/fmt/base.h:1383:19:   required from ‘fmt::v11::detail::value<Context>::value(T&) [with T = const boost::basic_format<char>; Context = fmt::v11::context]’
 1383 |     custom.format = format_custom_arg<
      |     ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
 1384 |         value_type, typename Context::template formatter_type<value_type>>;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/fmt/base.h:1647:41:   required from ‘constexpr fmt::v11::detail::value<Context> fmt::v11::detail::make_arg(T&) [with bool PACKED = true; Context = fmt::v11::context; T = const boost::basic_format<char>; typename std::enable_if<PACKED, int>::type <anonymous> = 0]’
 1647 |   return {arg_mapper<Context>().map(val)};
      |                                         ^
/usr/include/fmt/base.h:2018:74:   required from ‘constexpr fmt::v11::detail::format_arg_store<Context, NUM_ARGS, 0, DESC> fmt::v11::make_format_args(T& ...) [with Context = context; T = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >}; long unsigned int NUM_ARGS = 1; long unsigned int NUM_NAMED_ARGS = 0; long long unsigned int DESC = 15; typename std::enable_if<(NUM_NAMED_ARGS == 0), int>::type <anonymous> = 0]’
 2018 |   return {{detail::make_arg<NUM_ARGS <= detail::max_packed_args, Context>(
      |            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
 2019 |       args)...}};
      |       ~~~~~                                                               
/usr/include/spdlog/logger.h:328:75:   required from ‘void spdlog::logger::log_(spdlog::source_loc, spdlog::level::level_enum, spdlog::string_view_t, Args&& ...) [with Args = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&}; spdlog::string_view_t = fmt::v11::basic_string_view<char>]’
  328 |             fmt::vformat_to(fmt::appender(buf), fmt, fmt::make_format_args(args...));
      |                                                      ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
/usr/include/spdlog/logger.h:80:13:   required from ‘void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, fmt::v11::format_string<T ...>, Args&& ...) [with Args = {const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&}; fmt::v11::format_string<T ...> = fmt::v11::basic_format_string<char, const boost::basic_format<char, std::char_traits<char>, std::allocator<char> >&>]’
   80 |         log_(loc, lvl, details::to_string_view(fmt), std::forward<Args>(args)...);
      |         ~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/spdlog/logger.h:98:12:   required from ‘void spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, const T&) [with T = boost::basic_format<char>; typename std::enable_if<(! spdlog::is_convertible_to_any_format_string<const T&>::value), int>::type <anonymous> = 0]’
   98 |         log(loc, lvl, "{}", msg);
      |         ~~~^~~~~~~~~~~~~~~~~~~~~
/usr/include/spdlog/logger.h:90:12:   required from ‘void spdlog::logger::log(spdlog::level::level_enum, const T&) [with T = boost::basic_format<char>]’
   90 |         log(source_loc{}, lvl, msg);
      |         ~~~^~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/spdlog/logger.h:253:12:   required from ‘void spdlog::logger::error(const T&) [with T = boost::basic_format<char>]’
  253 |         log(level::err, msg);
      |         ~~~^~~~~~~~~~~~~~~~~
/home/mock/trunk-recorder/lib/op25_repeater/lib/iqfile_source_impl.cc:188:9:   required from here
  188 |         GR_LOG_ERROR(d_logger, boost::format("%s: %s") % filename % strerror(errno));
      |         ^
/usr/include/fmt/base.h:1402:29: error: passing ‘const fmt::v11::formatter<boost::basic_format<char> >’ as ‘this’ argument discards qualifiers [-fpermissive]
 1402 |     ctx.advance_to(cf.format(*static_cast<qualified_type*>(arg), ctx));
      |                    ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/gnuradio/custom_lock.h:15,
                 from /usr/include/gnuradio/buffer.h:15,
                 from /usr/include/gnuradio/buffer_double_mapped.h:15,
                 from /usr/include/gnuradio/io_signature.h:29:
/usr/include/gnuradio/logger.h:323:10: note:   in call to ‘decltype (((fmt::v11::formatter<boost::basic_format<char> >*)this)->fmt::v11::detail::native_formatter<fmt::v11::basic_string_view<char>, char, fmt::v11::detail::type::string_type>::format<FormatContext>(bfmt.boost::basic_format<char>::str(), ctx)) fmt::v11::formatter<boost::basic_format<char> >::format(const boost::format&, FormatContext&) [with FormatContext = fmt::v11::context; decltype (((fmt::v11::formatter<boost::basic_format<char> >*)this)->fmt::v11::detail::native_formatter<fmt::v11::basic_string_view<char>, char, fmt::v11::detail::type::string_type>::format<FormatContext>(bfmt.boost::basic_format<char>::str(), ctx)) = fmt::v11::basic_appender<char>; boost::format = boost::basic_format<char>]’
  323 |     auto format(const boost::format& bfmt, FormatContext& ctx)
      |          ^~~~~~

taclane, 2024-09-27:
If you have a moment, could you try a quick fix?

In `lib/op25_repeater/lib/iqfile_source_impl.cc`, starting around line 185, adding parentheses around the boost formatting statement, and returning it as a string should be all that's needed.

```diff
@@ -185,14 +185,14 @@ void iqfile_source_impl::open(const char* filename,
     }

     if ((d_new_fp = fopen(filename, "rb")) == NULL) {
-        GR_LOG_ERROR(d_logger, boost::format("%s: %s") % filename % strerror(errno));
+        GR_LOG_ERROR(d_logger, (boost::format("%s: %s") % filename % strerror(errno)).str());
         throw std::runtime_error("can't open file");
     }

     struct GR_STAT st;

     if (GR_FSTAT(GR_FILENO(d_new_fp), &st)) {
-        GR_LOG_ERROR(d_logger, boost::format("%s: %s") % filename % strerror(errno));
+        GR_LOG_ERROR(d_logger, (boost::format("%s: %s") % filename % strerror(errno)).str());
         throw std::runtime_error("can't fstat file");
     }
     if (S_ISREG(st.st_mode)) {
```

This was similarly affecting MacOS builds as of a couple weeks ago, and the above addresses the fmt/spdlog update on that platform (homebrew).  If that seems to do the trick for you, I can put in a PR.

mldulaney, 2024-09-29:
Yeah, this did the trick

taclane, 2024-09-29:
Great! The fix has been submitted; hopefully it can get merged in soon.

robotastic, 2024-10-01:
Awesome work - thanks for tracking this down!

closed 2024-10-01 by robotastic