TrunkRecorder/trunk-recorder #980: Segmentation fault (core dumped) ... and getting discouraged

thedanielhunt, 2024-10-27:
Been working in earnest on my build with two RTL-SDRs, and the system keeps dumping out. I know many will say "turn off AGC" but that seems to be the only way to get signals captured. Anyone got any guidance on what I'm doing wrong?

`{
  "ver": 2,
  "frequencyFormat": "mhz",
  "sources": [
  {
      "center": 853244000,
      "rate": 1512000,
      "gain": 7.7,
      "digitalRecorders": 16,
      "analogRecorders": 0,
      "driver": "osmosdr",
      "device": "rtl=0",
      "agc": true,
      "ppm": -1.5
    },
      {
      "center": 851732000,
      "rate": 1512000,
      "gain": 7.7,
      "digitalRecorders": 16,
      "analogRecorders": 0,
      "driver": "osmosdr",
      "device": "rtl=1",
      "agc": true,
      "ppm": -3.0
    }
  ],
  "systems": [
    {
      "shortName": "srrcs",
      "talkgroupsFile": "talkgroups.csv",
      "talkgroupDisplayFormat": "id_tag",
      "modulation": "qpsk",
      "squelch": -70,
      "control_channels": [
        853450000,
        852350000,
        853187500,
        853875000
      ],
      "channels": [
        851700000,
        851050000,
        851262500,
        851325000,
        851387500,
        851437500,
        851600000,
        851625000,
        851650000,
        851675000,
        851750000,
        851800000,
        851825000,
        851850000,
        852075000,
        852250000,
        852300000,
        852462500,
        852575000,
        852687500,
        852737500,
        853162500,
        853262500,
        853612500,
        853712500,
        853800000,
        853900000
[... 71 lines cut ...]
[2024-10-27 15:29:12.011141] (info)   	 - System Source 0 - Min Freq: 852.488031 MHz Max Freq: 853.999969 MHz
[2024-10-27 15:29:12.031010] (info)   	 Xlating Channelizer decimator - freq_xlating taps: 57 Decim: 4 Resampled Rate: 25200 Lowpass Taps: 87
[2024-10-27 15:29:12.031051] (info)   	 Channelizer ARB - Symbol Rate: 24000 Resampled Rate: 25200 ARB Rate: 0.952381 ARB Taps: 229 BW: 0.380952 TB: 0.190476
Allocating 15 zero-copy buffers
Allocating 15 zero-copy buffers
[2024-10-27 15:29:12.285771] (error)   [srrcs]	 Control Channel Message Decode Rate: 0/sec, count:  1
[2024-10-27 15:29:13.036978] (error)   P25 Parse error, s:  Len: 0
[2024-10-27 15:29:14.084751] (error)   P25 Parse error, s:  Len: 0
[2024-10-27 15:29:15.019605] (error)   [srrcs] Retuning to Control Channel: 853.875000 MHz
[2024-10-27 15:29:15.019644] (info)   	 - System Source 0 - Min Freq: 852.488031 MHz Max Freq: 853.999969 MHz
[2024-10-27 15:29:15.019661] (error)   [srrcs]	 Control Channel Message Decode Rate: 0/sec, count:  1
[2024-10-27 15:29:15.121732] (error)   P25 Parse error, s:  Len: 0
[2024-10-27 15:29:16.159156] (error)   P25 Parse error, s:  Len: 0
[2024-10-27 15:29:17.197342] (error)   P25 Parse error, s:  Len: 0
[2024-10-27 15:29:18.019344] (error)   [srrcs] Retuning to Control Channel: 853.450000 MHz
[2024-10-27 15:29:18.019384] (info)   	 - System Source 0 - Min Freq: 852.488031 MHz Max Freq: 853.999969 MHz
[2024-10-27 15:29:18.019401] (error)   [srrcs]	 Control Channel Message Decode Rate: 0/sec, count:  1
[2024-10-27 15:29:18.245281] (error)   P25 Parse error, s:  Len: 0
[2024-10-27 15:29:19.714828] (error)   P25 Parse error, s:  Len: 0
[2024-10-27 15:29:20.750708] (error)   P25 Parse error, s:  Len: 0
[2024-10-27 15:29:21.018048] (error)   [srrcs] Retuning to Control Channel: 852.350000 MHz
[2024-10-27 15:29:21.018086] (info)   	 - System Source 1 - Min Freq: 850.976031 MHz Max Freq: 852.487969 MHz
[2024-10-27 15:29:21.038560] (info)   	 Xlating Channelizer decimator - freq_xlating taps: 57 Decim: 4 Resampled Rate: 25200 Lowpass Taps: 87
[2024-10-27 15:29:21.038626] (info)   	 Channelizer ARB - Symbol Rate: 24000 Resampled Rate: 25200 ARB Rate: 0.952381 ARB Taps: 229 BW: 0.380952 TB: 0.190476
Segmentation fault (core dumped)`

thedanielhunt, 2024-10-27:
And this what the system is doing when it dumps out: 

```
[2024-10-27 15:51:37.250481] (info)   [srrcs]	31C	TG:       3565 (                      -)	Freq: 851.825000 MHz	Not Recording: TG not in Talkgroup File 
[2024-10-27 15:51:37.691984] (info)   [srrcs]	32C	TG:       3849 (                  SSD 2)	Freq: 852.250000 MHz	Starting P25 Recorder Num [16]	TDMA: false	Slot: 0	QPSK: true
[2024-10-27 15:51:37.692020] (info)   [srrcs]	32C	TG:       3849 (                  SSD 2)	Freq: 852.250000 MHz	This was an UPDATE
[2024-10-27 15:51:39.001593] (info)   [srrcs]	29C	TG:       3037 (                CH PD 2)	Freq: 853.612500 MHz	Concluding Recorded Call - Last Update: 4s	Recorder last write:4.78425	Call Elapsed: 7
[2024-10-27 15:51:39.001623] (info)   [srrcs]	29C	TG:       3037 (                CH PD 2)	Freq: 853.612500 MHz	Call was UPDATE not GRANT
[2024-10-27 15:51:39.001658] (info)   [srrcs]	29C	TG:       3037 (                CH PD 2)	Freq: 853.612500 MHz	Stopping P25 Recorder Num [0]	TDMA: false	Slot: 0	Hz Error: -434
[2024-10-27 15:51:39.001708] (info)   [srrcs]	29C	TG:       3037 (                CH PD 2)	Freq: 853.612500 MHz	- Transmission src: 9020135 pos:  0.00 length:  1.62 errors: 35 spikes: 6
[2024-10-27 15:51:40.014728] (error)   [srrcs]	 Control Channel Message Decode Rate: 9/sec, count:  28
[2024-10-27 15:51:41.242948] (info)   [srrcs]	32C	TG:       3849 (                  SSD 2)	Freq: 852.250000 MHz	Unit ID set via Control Channel, ext: 9180551	current: -1	 samples: 0
[2024-10-27 15:51:42.010020] (info)   [srrcs]	30C	TG:       3277 (                EG PD 1)	Freq: 853.800000 MHz	Concluding Recorded Call - Last Update: 5s	Recorder last write:3.37179	Call Elapsed: 10
[2024-10-27 15:51:42.010049] (info)   [srrcs]	30C	TG:       3277 (                EG PD 1)	Freq: 853.800000 MHz	Call was UPDATE not GRANT
[2024-10-27 15:51:42.010063] (info)   [srrcs]	30C	TG:       3277 (                EG PD 1)	Freq: 853.800000 MHz	Stopping P25 Recorder Num [1]	TDMA: false	Slot: 0	Hz Error: -342
[2024-10-27 15:51:42.010128] (info)   [srrcs]	30C	TG:       3277 (                EG PD 1)	Freq: 853.800000 MHz	- Transmission src: 9180101 pos:  0.00 length:  3.78 errors: 327 spikes: 20
[2024-10-27 15:51:42.010142] (info)   [srrcs]	30C	TG:       3277 (                EG PD 1)	Freq: 853.800000 MHz	- Transmission src: 9180101 pos:  3.78 length:  1.62 errors: 224 spikes: 11
[2024-10-27 15:51:43.012936] (error)   [srrcs]	 Control Channel Message Decode Rate: 5/sec, count:  15
[2024-10-27 15:51:43.146050] (info)   [srrcs]	33C	TG:       3565 (                      -)	Freq: 851.825000 MHz	Not Recording: TG not in Talkgroup File 
[2024-10-27 15:51:43.146113] (info)   [srrcs]	34C	TG:       3037 (                CH PD 2)	Freq: 851.325000 MHz	Starting P25 Recorder Num [17]	TDMA: false	Slot: 0	QPSK: true
[2024-10-27 15:51:43.146144] (info)   [srrcs]	34C	TG:       3037 (                CH PD 2)	Freq: 851.325000 MHz	This was an UPDATE
[2024-10-27 15:51:43.146226] (info)   [srrcs]	35C	TG:       3851 (                  SSD 3)	Freq: 853.900000 MHz	Starting P25 Recorder Num [0]	TDMA: false	Slot: 0	QPSK: true
[2024-10-27 15:51:43.146278] (info)   [srrcs]	35C	TG:       3851 (                  SSD 3)	Freq: 853.900000 MHz	This was an UPDATE
[2024-10-27 15:51:44.186631] (error)   P25 Parse error, s:  Len: 0
[2024-10-27 15:51:45.227122] (error)   P25 Parse error, s:  Len: 0
[2024-10-27 15:51:46.003227] (info)   [srrcs]	32C	TG:       3849 (                  SSD 2)	Freq: 852.250000 MHz	Concluding Recorded Call - Last Update: 5s	Recorder last write:3.31229	Call Elapsed: 9
[2024-10-27 15:51:46.003273] (info)   [srrcs]	32C	TG:       3849 (                  SSD 2)	Freq: 852.250000 MHz	Call was UPDATE not GRANT
[2024-10-27 15:51:46.003306] (info)   [srrcs]	32C	TG:       3849 (                  SSD 2)	Freq: 852.250000 MHz	Stopping P25 Recorder Num [16]	TDMA: false	Slot: 0	Hz Error: 1044
[2024-10-27 15:51:46.003432] (info)   [srrcs]	32C	TG:       3849 (                  SSD 2)	Freq: 852.250000 MHz	- Transmission src: 9101979 pos:  0.00 length:  2.34 errors: 33 spikes: 9
[2024-10-27 15:51:46.003465] (info)   [srrcs]	32C	TG:       3849 (                  SSD 2)	Freq: 852.250000 MHz	- Transmission src: 9180551 pos:  2.34 length:  1.08 errors: 32 spikes: 4
[2024-10-27 15:51:46.013695] (error)   [srrcs] Retuning to Control Channel: 852.350000 MHz
[2024-10-27 15:51:46.013825] (info)   	 - System Source 1 - Min Freq: 850.976031 MHz Max Freq: 852.487969 MHz
[2024-10-27 15:51:46.035999] (info)   	 Xlating Channelizer decimator - freq_xlating taps: 57 Decim: 4 Resampled Rate: 25200 Lowpass Taps: 87
[2024-10-27 15:51:46.036124] (info)   	 Channelizer ARB - Symbol Rate: 24000 Resampled Rate: 25200 ARB Rate: 0.952381 ARB Taps: 229 BW: 0.380952 TB: 0.190476
Segmentation fault (core dumped)

```

frameshift18, 2024-10-27:
That decode rate is very poor, I can't imagine that's helping things. I'd try to improve your received signal strength first. Even if it didn't seg fault your decoded audio can't sound good at all.

thedanielhunt, 2024-10-27:
> That decode rate is very poor, I can't imagine that's helping things. I'd try to improve your received signal strength first. Even if it didn't seg fault your decoded audio can't sound good at all.

I'm going to switch to a beefer antenna and see if that helps.

tadscottsmith, 2024-10-27:
Are you on v5.x? 32 recorders is going to take a good chunk of CPU. Your best bet is to run some sort of graphical tool to visualize your control channel, get the correct gain identified, and turn off AGC. If you can not do that, I'd suggest taking out the `"squelch": -70,` (squelch is not needed for P25 trunked) and reduce the number of digital recorders down to 1 or 2 per source just so you can rule out any sort of hardware bottleneck. If you can get the signal dialed in, then you can add more recorders back into the equation.

frameshift18, 2024-10-27:
I suggest to look at the control channel through a SDR program. Something like SDR++. You can tell a lot about how things are looking and can tweak your SDR settings and get realtime feedback.

thedanielhunt, 2024-10-27:
Thank you for these suggestions. I just switched from my Log window antenna to the beefy dipole on the roof. I've been tinkering and tinkering for six weeks on this -- switching between GQRX and the command line -- and trying things here and there as I read the documentation over and over. 

I had a solid head-start from the OpenMhz contributor in my area but I didn't want to bog him down with dumb questions. 

Thank you, all, for the support and kind words.

thedanielhunt, 2024-10-27:
> Are you on v5.x? 32 recorders is going to take a good chunk of CPU. Your best bet is to run some sort of graphical tool to visualize your control channel, get the correct gain identified, and turn [...]

Yep, got 5.0 and taking the number down got everything a little more stable. I'm gonna try SDR++ too and see if that helps me get this dialed in.

The funny thing: I've been listening to this trunking system for years using that window antenna and a bank of Uniden Bearcats and never had an issue. I had no idea the signal was so weak.

Thanks again to all who lent me a hand.

closed 2024-10-27 by thedanielhunt

tadscottsmith, 2024-10-27:
Some of the error numbers are a little misleading. 

When you see something like `4.68 errors: 38 spikes: 18` it is just telling you the number of P25 errors that were corrected. P25 dedicates almost 30% of the data that it transfers to error correction, so your numbers are not alarming to me.

When you see something like `Slot: 0	Hz Error: 9`, that means that trunk-recorder found the actual signal 9hz away away from the channel that was advertised. At 850 MHz, any Hz Error +/- 500 Hz is pretty acceptable. If these are consistently higher or lower, this is where you might need to adjust your PPM to offset this error.

If you want to cleanup your config even further, you don't need the `channels` section for P25 Trunked. It's probably not hurting anything to have them in there, but the program will get the full list of channels from the control channel.

If you're a Discord guy, it's a very helpful community and I know there are others in your area monitoring the same system. https://discord.gg/btJAhESnks

thedanielhunt, 2024-10-27:
> If you're a Discord guy, it's a very helpful community and I know there are others in your area monitoring the same system. https://discord.gg/btJAhESnks

Boom, I'm there. Appreciate the advice, too. I have a lot to learn!