TrunkRecorder/trunk-recorder #981: Stability request -- safely handle a disconnecting / flaky RTL-SDR

jackdpeterson, 2024-10-29:
I'm just getting into playing with the whole RTL-SDR side of things and something I've noticed is that these little devices are nifty; but the software around using them (and configuring them) is fairly complex and non-resilient to hardware issues.

One thing that I haven't figured out is why my server will suddenly stop exposing the USB-Device to the virtual machine running Ubuntu 24.04 (resulting in trunk-recorder not running / crashing / not starting).

In the scenario where it's not starting ... it's because there isn't an SDR that exists under that ID or serial number (this makes sense); e.g., `rtl_test` only shows 2 devices and not 3. But ... it'd be nice if it just started up, threw an ERROR or CRIT level log and continued on and then just failed to scan those particular frequencies that the given SDR covers.

In the scenario where the SDR gets disconnected / bumped ... it'd be nice if the same behavior happened there -- throw a CRIT / ERROR log, stop the corresponding threads (analog / digital recorders), and then complain whenever a frequency needs to go to an uncovered bit of spectrum. 

**Hardware Set-up:**
HP DL325 Gen 10 -- has two USB-A connections in the back and 1x USB-A connection on the front. I'm using all three (each with a Nooelec rtl-sdr (SMART v5).

**Software**
VMWare ESXi 7.0
Virtual Machine (ubuntu 24.04) w/ each USB device being passed into the VM.

ZeroChaos-, 2024-10-29:
dupe of https://github.com/robotastic/trunk-recorder/issues/644

robotastic, 2025-01-10:
I am going to close this and track in #644

closed 2025-01-10 by robotastic