TrunkRecorder/trunk-recorder #988: Stats Socket breaks with Boost 1.87

robotastic, 2024-12-18:
The Stats Socket plugin stops working with Boost 1.87. The WebSocket library being used ( https://github.com/zaphoyd/websocketpp ) hasn't been updated in 4 years, so it is not likely this will be fixed soon. The best option is to probably pull Stats Socket from the build for now until a new library can be slotted in.

Issue from Websocketpp repo:
https://github.com/zaphoyd/websocketpp/issues/1157

robotastic, 2024-12-18:
Example of the build error:

```bash
[ 93%] Building CXX object plugins/stat_socket/CMakeFiles/stat_socket.dir/stat_socket.cc.o
In file included from /Users/luke/Projects/TrunkRecorder/trunk-recorder/plugins/stat_socket/stat_socket.cc:6:
In file included from /Users/luke/Projects/TrunkRecorder/trunk-recorder/lib/websocketpp/config/asio_no_tls_client.hpp:32:
In file included from /Users/luke/Projects/TrunkRecorder/trunk-recorder/lib/websocketpp/transport/asio/endpoint.hpp:32:
/Users/luke/Projects/TrunkRecorder/trunk-recorder/lib/websocketpp/transport/asio/connection.hpp:89:24: error: no type named 'io_service' in namespace 'websocketpp::lib::asio'
   89 |     typedef lib::asio::io_service * io_service_ptr;
      |             ~~~~~~~~~~~^
/Users/luke/Projects/TrunkRecorder/trunk-recorder/lib/websocketpp/transport/asio/connection.hpp:91:40: error: no member named 'io_service' in namespace 'websocketpp::lib::asio'
   91 |     typedef lib::shared_ptr<lib::asio::io_service::strand> strand_ptr;
      |                             ~~~~~~~~~~~^
/Users/luke/Projects/TrunkRecorder/trunk-recorder/lib/websocketpp/transport/asio/connection.hpp:315:17: error: no matching constructor for initialization of 'lib::asio::steady_timer' (aka 'basic_waitable_timer<chrono::steady_clock>')
  315 |             new lib::asio::steady_timer(
      |                 ^
  316 |                 *m_io_service,
      |                 ~~~~~~~~~~~~~~
  317 |                 lib::asio::milliseconds(duration))
  ```

miyanyan, 2024-12-19:
asio removed some previously deprecated facilities, see https://www.boost.org/doc/libs/1_87_0/doc/html/boost_asio/history.html, this's my patch https://github.com/microsoft/vcpkg/pull/42678/files#diff-359671c6dbccd346495bb59cd5f29605921ff4c4448e6a136a89d873438b0c60

aaknitt, 2024-12-21:
I think I'll also need to update the simplestream plugin due to this issue.  Here's a dump from GPT that explains what needs to change.  I'll put this on my to-do list for simplestream and the same changes should be able to be applied to this plugin as well, I think.  

> In some c++ code I'm getting this error after updating to a newer version of Boost.  I think this might be because of some deprecated stuff in asio.  how can I fix it?

```
simplestream.cc:52:63: error: no member named 'from_string' in 'boost::asio::ip::address'
   52 |       stream.remote_endpoint = ip::udp::endpoint(ip::address::from_string(stream.address), stream.port);
```

> ChatGPT said:
> ChatGPT
> The error occurs because boost::asio::ip::address::from_string has been deprecated and removed in newer versions of Boost.Asio. Instead, you should use boost::asio::ip::make_address, which is the modern replacement for creating an ip::address from a string.
> 
> Updated Code
> 
> Replace:

`ip::address::from_string(stream.address)`

> With:

`boost::asio::ip::make_address(stream.address)`

> So the full line becomes:

`stream.remote_endpoint = ip::udp::endpoint(boost::asio::ip::make_address(stream.address), stream.port);`

> Why This Works
> boost::asio::ip::make_address is the new function introduced to replace from_string. It is more consistent with modern C++ practices and aligns with the general direction of Boost.Asio.
> Additional Notes
> Error Handling:
> 
> boost::asio::ip::make_address may throw an exception if the address string is invalid. You can catch this exception and handle it appropriately.

```
try {
    auto address = boost::asio::ip::make_address(stream.address);
    stream.remote_endpoint = ip::udp::endpoint(address, stream.port);
} catch (const std::exception& e) {
    std::cerr << "Invalid address: " << stream.address << " - " << e.what() << std::endl;
}
```

> Support for IPv4 and IPv6:
> 
> boost::asio::ip::make_address supports both IPv4 and IPv6 addresses, so your code will work seamlessly for either.
> Documentation Reference:
> 
> You can refer to the Boost.Asio documentation for more details about make_address: Boost.Asio Documentation.
> Conclusion
> The boost::asio::ip::make_address function is a direct replacement for the deprecated boost::asio::ip::address::from_string. Update your code to use it, and handle exceptions if needed to ensure robust error handling for invalid addresses. Let me know if you encounter additional issues! 🚀

> You said:
> what about these?

```
/Users/me/src/sdr/trunk-recorder/plugins/simplestream/simplestream.cc:13:1: error: unknown type name 'io_service'
   13 | io_service my_tcp_io_service;
      | ^
/Users/me/src/sdr/trunk-recorder/plugins/simplestream/simplestream.cc:36:24: error: no type named 'io_service' in namespace 'boost::asio'
   36 |   typedef boost::asio::io_service io_service;
      |           ~~~~~~~~~~~~~^
/Users/me/src/sdr/trunk-recorder/plugins/simplestream/simplestream.cc:39:28: error: no matching constructor for initialization of 'ip::udp::socket' (aka 'basic_datagram_socket<udp>')
   39 |   ip::udp::socket my_socket{my_io_service};
```

> ChatGPT said:
> ChatGPT
> These errors indicate that the io_service type in Boost.Asio has been replaced with io_context in newer versions of Boost. The io_context class serves the same purpose as io_service but was introduced to better align with modern C++ practices.
> 
> Fixes for Each Error
> 1. Replace io_service with io_context
> Change all instances of io_service to io_context:

`boost::asio::io_service my_tcp_io_service;`

> Becomes:

`boost::asio::io_context my_tcp_io_context;`

> 2. Update typedef for io_service
> If you’re typedefing io_service, update it to use io_context:

`typedef boost::asio::io_service io_service;`

> Becomes:
`typedef boost::asio::io_context io_context;`

> 3. Update ip::udp::socket Constructor
> The ip::udp::socket constructor expects a reference to an io_context object instead of an io_service. Replace:

`ip::udp::socket my_socket{my_io_service};`

> With:

`ip::udp::socket my_socket{my_tcp_io_context};`

> Complete Updated Code Snippet
> Here’s how the updated code might look:

#include <boost/asio.hpp>

```
namespace ip = boost::asio::ip;

int main() {
    // Create an io_context instance
    boost::asio::io_context my_tcp_io_context;

    // Create a UDP socket using the io_context
    ip::udp::socket my_socket{my_tcp_io_context};

    // Example usage
    ip::udp::endpoint endpoint(ip::make_address("127.0.0.1"), 12345);
    my_socket.open(endpoint.protocol());
    my_socket.bind(endpoint);

    return 0;
}
```

> Why These Changes Are Necessary
> io_service Deprecated:
> 
> boost::asio::io_service has been deprecated and removed in favor of boost::asio::io_context.
> Consistent API:
> 
> Constructors like ip::udp::socket and other Asio components now expect io_context references instead of io_service.
> Additional Tips
> If your code uses asynchronous operations, ensure you call my_tcp_io_context.run() to process events in the io_context.
> Summary of Changes
> Replace io_service with io_context.
> Update typedefs and variable declarations accordingly.
> Ensure components like ip::udp::socket use the updated io_context.
> With these updates, your code should be compatible with newer versions of Boost.Asio. Let me know if you encounter further issues! 🚀

jcelerier, 2024-12-27:
If that can be useful my fork updates the lib to work with boost 1.87+ : https://github.com/jcelerier/websocketpp/tree/ossia/2024-12-18

robotastic, 2024-12-31:
Thanks @jcelerier  @miyanyan - I think I might just remove this portion of the code and move to using MQTT

andyreagan, 2025-01-03:
Fwiw I'm seeing these errors and unable to build on OSX (with boost 1.76 or 1.85, the options from brew).

robotastic, 2025-01-03:
I am going to pull the parts out and make them into a separate repo for that plugin... change coming soon.

closed 2025-01-05 by robotastic

israelgoldstein, 2025-01-23:
is this the same reason simplestream does not work ?

PR #991 (2025-01-05): Dev/backlog
Fixes #988 by removing Stat Socket plugin to its own repo, here: https://github.com/TrunkRecorder/trunk-recorder-stat-socket

Also fixes #379 and adds Tag to the .m4a files that get created.