TrunkRecorder/trunk-recorder #998: invalid signal and noise readings

ZeroChaos-, 2025-02-03:
I'm getting json files with signal or noise as 999.  I suspect something is wrong, but it's difficult to tell what based on the available information.

Examples:

```
{
  "freq": 460125000,
  "freq_error": -31,
  "signal": 999,
  "noise": -50,
  "source_num": 3,
  "recorder_num": 42,
  "tdma_slot": 0,
  "phase2_tdma": 0,
  "start_time": 1738480876,
  "stop_time": 1738480879,
  "emergency": 0,
  "priority": 0,
  "mode": 0,
  "duplex": 0,
  "encrypted": 0,
  "call_length": 1,
  "talkgroup": 170,
  "talkgroup_tag": "West PD S-2(Coraopolis)",
  "talkgroup_description": "",
  "talkgroup_group_tag": "Police Dispatch",
  "talkgroup_group": "Allegheny County",
  "audio_type": "analog",
  "short_name": "conv",
  "freqList": [
    {
      "freq": 460125000,
      "time": 1738480876,
      "pos": 0.0,
      "len": 1.0439375,
      "error_count": 50,
      "spike_count": 999
    }
  ],
  "srcList": [
    {
      "src": -1,
      "time": 1738480876,
      "pos": 0.0,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    }
  ]
}
```
```
{
  "freq": 470437500,
  "freq_error": -4,
  "signal": -56,
  "noise": 999,
  "source_num": 9,
  "recorder_num": 52,
  "tdma_slot": 0,
  "phase2_tdma": 0,
  "start_time": 1738510990,
  "stop_time": 1738511278,
  "emergency": 0,
  "priority": 0,
  "mode": 0,
  "duplex": 0,
  "encrypted": 0,
  "call_length": 180,
  "talkgroup": 99,
  "talkgroup_tag": "West EMS Dispatch Identified",
  "talkgroup_description": "",
  "talkgroup_group_tag": "EMS Dispatch",
  "talkgroup_group": "Allegheny Country",
  "audio_type": "analog",
  "short_name": "conv",
  "freqList": [
    {
      "freq": 470437500,
      "time": 1738510990,
      "pos": 0.0,
      "len": 180.0841875,
      "error_count": 999,
      "spike_count": 56
    }
  ],
  "srcList": [
    {
      "src": 4221,
      "time": 1738510990,
      "pos": 0.0,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    }
  ],
  "multimon_ng": []
}
```

(I add the multimon_ng key but I didn't manipulate the noise or signal)

ZeroChaos-, 2025-02-04:
Given the issue reported is for signal and noise of 999 my suspicion is the opposite of yours.  I suspect that it an initialization value and the real value was never set for some reason.

tadscottsmith, 2025-02-27:
I'm not sure yet why they aren't always getting set to but the 999 is coming from DB_UNSET which is the default value when the recorders start/restart.

https://github.com/robotastic/trunk-recorder/blob/469744619aaaebe645712f27a5d87753fbbccbfd/trunk-recorder/global_structs.h#L8

xmleger, 2025-03-03:
OK... the -99 idea did not work.