TrunkRecorder/trunk-recorder #1009: Broadcastify Plugin package errors

jared-bloomer, 2025-03-14:
When using the Broadcastify Plugin I am getting the following error

```
trunk-recorder  | [2025-03-14 15:38:08.014232] (info)   Combining: /dev/shm/Henry County (P25) (Simulcast)/337-1741981086_853875000.0.wav  into: /app/media/Henry County (P25) (Simulcast)/2025/3/14/337-1741981086_853875000.0-call_22.wav
trunk-recorder  | [2025-03-14 15:38:08.014361] (info)   sox /dev/shm/Henry County (P25) (Simulcast)/337-1741981086_853875000.0.wav  /app/media/Henry County (P25) (Simulcast)/2025/3/14/337-1741981086_853875000.0-call_22.wav
trunk-recorder  | [2025-03-14 15:38:08.014382] (error)   Failed to combine recordings, see above error. Make sure you have sox and fdkaac installed.
```

I have opened Pull Request #1008 to ensure these packages are installed. I have them installed in the image I am using

```
root@8fff3466f798:/app# apt install fdkaac
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
fdkaac is already the newest version (1.0.0-1build1).
0 upgraded, 0 newly installed, 0 to remove and 40 not upgraded.
root@8fff3466f798:/app# apt install sox
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
sox is already the newest version (14.4.2+git20190427-2+deb11u2ubuntu0.22.04.1).
0 upgraded, 0 newly installed, 0 to remove and 40 not upgraded.
root@8fff3466f798:/app#
```

Not sure why this error is being triggered if the packages are installed.

jared-bloomer, 2025-03-14:
This was an issue related to the name i defined in my `short_name` in my config.json

closed 2025-03-14 by jared-bloomer

taclane, 2025-03-14:
Actually, if you want to reopen this, it's something that's cropped up before.  

When audio is combined and converted, filenames should be escaped to account for any potential spaces.