TrunkRecorder/trunk-recorder #1010: Broadcastify Plugin - INVALID-SYSTEM-ID

jared-bloomer, 2025-03-14:
I am getting the following error in my logs

```
trunk-recorder  | [2025-03-14 16:04:11.393790] (error)   [Henry_County_P25_Simulcast]   19C     TG:        273  Freq: 853.212500 MHz    Broadcastify Metadata Upload Error: INVALID-SYSTEM-ID
trunk-recorder  | [2025-03-14 16:04:11.395224] (error)   [Henry_County_P25_Simulcast]   19C     TG:        273  Freq: 853.212500 MHz    Plugin Manager: call_end -  broadcastify_uploader failed.
```

I have the system ID in config.json configured with the one Broadcastify gave me. 

I have noticed is in the [official documentation](https://wiki.radioreference.com/index.php/Broadcastify-Calls-Trunk-Recorder_P25_Federal_FBI_%C2%A7_DEA_P25_digital_Trunking_Scanner_Surveillance_Springfield_Division) they quote the ID in double quotes (i.e. `"broadcastifySystemId": "91",`). If I do that in my json config I get a type error about it needing to be an int and not a string. So, unfortunately I am not quoting the ID in my config. I am willing to bet this is related somehow. Without knowing where to look it is hard for me to track this down.

taclane, 2025-03-14:
The system ID should not be a quoted string in the trunk-recorder config.  The info on radioreference is maybe almost two years out of date at this point.

Additionally, if you want to detangle some stuff from your config, you can define the broadcastify plugin separately in the plugin section like:

```json
        "plugins": [
                {
                        "name": "Broadcastify",
                        "library": "libbroadcastify_uploader.so",
                        "enabled": true,
                        "broadcastifyCallsServer": "https://api.broadcastify.com/call-upload",
                        "systems": [
                                {
                                        "shortName": "tr_shortname_to_upload",
                                        "broadcastifySystemId": 1234,
                                        "broadcastifyApiKey": "bbac55c9-a1f0-44b8-968d-notarealkey1"
                                }
                        ]
                }
        ]
```

taclane, 2025-11-13:
The linked broadcastify wiki should now correctly show that the `broadcastifySystemId` key in the config uses an integer value, not a quoted number.

closed 2025-11-13 by taclane