TrunkRecorder/trunk-recorder #1025: trunk-recorder-status-server no longer working

theficus, 2025-06-14:
Not sure if this should go here or trunk-recorder-status-server, but I've found that with recent builds that nothing is being written to the trunk-recorder-status-server website anymore. I can see the connection being made, but nothing in the site ever seems to render.

I'm using the latest sources for both. (I realize that trunk-recorder-status-server hasn't been updated in quite some time.)

SimonSheehan, 2025-06-14:
A related library broke with a Boost update. It was subsequently removed https://github.com/TrunkRecorder/trunk-recorder/issues/988

theficus, 2025-06-17:
@SimonSheehan Is there any workaround? The linked issue is Closed.

theficus, 2025-06-17:
I have been using trunk-recorder-status-server. It's low effort and more than meets my needs for getting an at-a-glance status for my TR setup. 

I assume that even with the MQTT plugin I would also need some sort of front end for it?

bburkett52, 2026-05-31:
@theficus Do you, perhaps, have a docker image or docker-compose.yml example for this?  I'm pretty much a novice in all this! I am running T-R in docker currently.  Thank you!