TrunkRecorder/trunk-recorder #1028: SoX and fdkaac errors running on Docker

jj358mhz, 2025-07-10:
I am seeing this error on my Docker image running the latest 5.0.2 image:

`[2025-07-10 19:16:09.017340] (info)   sox /dev/shm/bascanner/5530-1752174960_771193750.1.wav  /app/bascanner/2025/7/10/5530-1752174960_771193750.1-call_49610.wav 
[2025-07-10 19:16:09.017355] (error)   Failed to combine recordings, see above error. Make sure you have sox and fdkaac installed.`

`pi@raspberrypi-scanner:~ $ sudo apt -y install libssl-dev openssl curl git fdkaac sox libcurl3-gnutls libcurl4 libcurl4-openssl-dev gnuradio gnuradio-dev gr-osmosdr libhackrf-dev libuhd-dev cmake make build-essential libboost-all-dev libusb-1.0-0-dev libsndfile1-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package fdkaac`

SoX is installed on my raspberry, but the fdkaac throws an error saying it cannot be found

Thoughts?

SimonSheehan, 2025-07-10:
Do you have the "contrib" component added to your apt SourceList?

jj358mhz, 2025-07-11:
@SimonSheehan So yes, I overlooked that part of the install. I added the Debian multimedia source, installed the SoX and fdkaac (as well as the others outlined in the instructions), then restarted my Docker container. It ran for about 2hrs, then went offline. This is an error snippet from the container log:

`[2025-07-11 01:24:34.009076] (error)   Failed to convert call recording, see above error. Make sure you have sox and fdkaac installed.`
`[2025-07-11 01:24:44.005153] (info)   [bascanner]	7000C	TG:      10005	Freq: 771.793750 MHz	Stopping P25 Recorder Num [9]	TDMA: true	Slot: 0	Hz Error: 1173`
`[2025-07-11 01:24:44.007248] (error)   Failed to combine recordings, see above error. Make sure you have sox and fdkaac installed.`
`ERROR: unsupported input file`
`[2025-07-11 01:24:44.009560] (error)   Failed to convert call recording, see above error. Make sure you have sox and fdkaac installed.`
`[2025-07-11 01:24:45.013265] (info)   [bascanner]	7001C	TG:       7505	Freq: 771.793750 MHz	Stopping P25 Recorder Num [10]	TDMA: true	Slot: 1	Hz Error: 937`
`[2025-07-11 01:24:45.015404] (error)   Failed to combine recordings, see above error. Make sure you have sox and fdkaac installed.`
`ERROR: unsupported input file`

robotastic, 2025-08-14:
Hmmm... I did a comparison and it looks like there were a lot of changes to the Docker build, but not a lot to TR itself. @mikecomer-beakon since you are building things yourself, can you try taking the Dockerfile from v5.0.1 and using it to build an image with TR checked out to v5.0.2?

Here is the v5.0.1 dockerfile: https://github.com/TrunkRecorder/trunk-recorder/blob/v5.0.1/Dockerfile

Comparison between versions:
https://github.com/TrunkRecorder/trunk-recorder/compare/v5.0.1...v5.0.2

mikecomer-beakon, 2025-08-14:
thank you @taclane and @robotastic.

Nick, I think you nailed it.   During the problem condition /dev/shm was indeed 100% in-use 0% available.   Thanks for your docker expertise - I changed it to 1gb and we already reboot daily - so I'm sure that will resolve this (for me).  I agree making it optional to save those files is a good idea.

[image]

taclane, 2025-08-15:
While I don't think this is a *"64MB ought to be enough for anybody!"* kind of thing, it should be still be able to buffer something like ~20 minutes of calls that are either active or waiting on upload.  But documenting that increasing the value on docker t-r with busy systems, or when you expect individual calls to have a long recording time, is probably very helpful.  

The docker-compose equivalent should look something along the lines of:
```yaml
services:
  recorder:
    image: robotastic/trunk-recorder:latest
    container_name: trunk-recorder
    shm_size: "1gb"  # increase "/dev/shm" space from default of 64 MB
    ...
```

But as far as the difference between 5.0.1 and 5.0.2, the only reason I can think of where the `TempDir` would inadvertantly fill up is #939.  That feature should probably be expanded on a little, such that it is an opt-in setting, and archives the failed uploads somewhere in `CaptureDir` (`CaptureDir/shortname/upload_failure`?) instead just leaving everything in-place where they could potentially fill tempfs.

taclane, 2025-08-22:
> another interesting thing ... after around 10 hours of monitoring about 35 conventional and convP25 channels ... I fill up to more than a hundred MB in shm. couple hundred MB by end of day. something makes this grow, though I'm not sure what failure accounts for it. I'm not sure I'm missing anything - just interesting that it fills over time.
> 

I'd double-check if you have a P25 or analog conventional channel that got stuck recording.  If squelch is set too low, or close enough to the noise floor, a recorder may inadvertently get triggered and attempt to continuously record dead air.

You can verify this by:
- During the periodic status display in the console, t-r lists which conventional channels are actively recording to disk.  Conventional channels should return to `State: idle` when they are not recording a call.
- Do a ctrl-C to trunk recorder while it is running, or gracefully stop it through other means.  It will attempt to close out the stuck recorder, and you may see a call with a particularly high `Call Elapsed:` value as it shuts down.
- `cd /dev/shm` and you should see directories with each of your system shortNames.  These are generally empty when nothing is actively happening.  The .wav files are normally just a couple hundred KB, and get deleted shortly after a call ends.  If you find .wavs that are in the tens of MB (or more), squelch for that system/channel may be too low.  The filename should give you some clues which recorder is getting stuck; it includes both the "talkgroup" identifier and what frequency the recorder is monitoring.