TrunkRecorder/trunk-recorder #1029: Crash with TR-4 and TR-5

lilhoser, 2025-07-20:
I'm on ubuntu 24.04 LTS-

```
# System Details Report
---

## Report details
- **Date generated:**                              2025-07-20 16:07:29

## Hardware Information:
- **Hardware Model:**                              Micro Computer _HK_ Tech Limited HX100G
- **Memory:**                                      64.0 GiB
- **Processor:**                                   AMD Ryzen™ 7 7840HS w/ Radeon™ 780M Graphics × 16
- **Graphics:**                                    AMD Radeon™ RX 6650M
- **Graphics 1:**                                  AMD Radeon™ 780M
- **Disk Capacity:**                               1.0 TB

## Software Information:
- **Firmware Version:**                            0.18
- **OS Name:**                                     Ubuntu 24.04.2 LTS
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               46
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.8.0-64-generic
```

I've been running TR-4 fine for over a year, and it started crashing after a recent system update. Coincidentally, I had also the same week checked the TR-5 repo out to try it, and it is also crashing. Here is the stack trace-

```
2025-07-20 16:05:11.012655] (info)   [whiteoakmt-hamilton]	7C	TG:       1137 (        CHA Police Disp)	Freq: 855.987500 MHz	Call length: 4.86 is less than min duration: 5
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO[2025-07-20 16:05:13.011141] (error)   [whiteoakmt-nbradley] Retuning to Control Channel: 769.606250 MHz
[2025-07-20 16:05:13.011513] (info)   	 - System Source 0 - Min Freq: 769.555625 MHz Max Freq: 771.475625 MHz
[2025-07-20 16:05:13.012006] (error)   [whiteoakmt-nbradley]	 Control Channel Message Decode Rate: 0/sec, count:  1
[2025-07-20 16:05:13.012431] (error)   Source 5 has stopped receiving samples - Terminating trunk recorder
terminate reached from thread id: 77fc65845c400x77fc6716993d: (gr::tagged_stream_block::check_topology(int, int)+0x142d)
0x77fc668bb0da: (std::rethrow_exception(std::__exception_ptr::exception_ptr)+0x8a)
0x77fc668a5a55: (std::terminate()+0x15)
0x77fc5cdb17c3: (std::thread::~thread()+0x25)
0x77fc66447a76: (erand48_r+0x1d6)
0x77fc66447bbe: (exit+0x1e)
0x60e81ce46397: (monitor_messages(Config&, std::shared_ptr<gr::top_block>&, std::vector<Source*, std::allocator<Source*> >&, std::vector<System*, std::allocator<System*> >&, std::vector<Call*, std::allocator<Call*> >&)+0xeb6)
0x60e81cd2b4c3: (main+0xa9b)
0x77fc6642a1ca: (__libc_init_first+0x8a)
0x77fc6642a28b: (__libc_start_main+0x8b)
0x60e81cd2a8a5: (_start+0x25)
Aborted (core dumped)
```

tadscottsmith, 2025-07-20:
A few things stand out to me.

The long string of `OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO` suggests an GNU Radio overflow condition, which typically indicates a resource bottleneck somewhere in the pipeline. This is most often caused by insufficient CPU resources or USB bandwidth limitations.

I also noticed it references source 5, which implies you're using at least six sources. Did you recently add any hardware that might increase USB usage? Are you using a powered USB hub? Unpowered hubs or shared USB buses can often not support that number of devices.

As for this actual crash, the logs show that t-r is intentionally terminating because source 5 stopped receiving samples. This is a new safety catch introduced in the later v5 versions. If you happen to have similar logs or a dump from v4, that could help determine whether this issue is new or just being logged differently now.

Without more specific details on your setup, I’d recommend first checking your USB bandwidth and power availability to see if you can see anything that might be reaching it's limits.

lilhoser, 2025-07-21:
Thanks for the info. Nothing about my USB setup has changed since I setup T4 about 2 years ago, and it has been working without any problem all this time. I'm using this 10-port Atolla hub Taclane uses, with 6 RTL-SDR v4's, exclusively for the SDRs - https://www.amazon.com/dp/B098KZMR4J

TR4 is also crashing, but I've not had any luck with a debug build and symbols. I will try to again tomorrow.

lilhoser, 2025-07-21:
The MiniForum's pc that runs this TR rig is dedicated to it 100% - the system is not doing anything else. I'm also not doing AI transcription on this system, but on another system completely independently. So I have no idea why it can't keep up. I will try to look more into this today.

robotastic, 2025-07-22:
It could also be worth try a "Performance" mode under power modes: https://askubuntu.com/questions/604720/setting-to-high-performance. It could help clear up the 'OOOOO'. It may also be worth reseating all of the USB drives, maybe one is loose. I have noticed that some SDRs can just fail over time. Add a small fan to cool the SDRs has helped for me.

One hack could just be setting up a Service to run Trunk Recorder. That way it should just restart if it ever crashes:
https://github.com/TrunkRecorder/trunk-recorder/blob/master/examples/SETUP_SERVICE.md

lilhoser, 2025-07-23:
Thanks @taclane !

closed 2025-07-23 by taclane