TrunkRecorder/trunk-recorder #1031: TrunkRecorder is not honoring the `blacklistTalkgroupsFile` option in my config.

MeckMyDay, 2025-07-26:
### Summary

TrunkRecorder is not honoring the `blacklistTalkgroupsFile` option in my config. I have confirmed the file is correctly mapped and formatted, but calls for blacklisted TGIDs (CATS talkgroups) are still being recorded.

### System Details

- **TrunkRecorder Version:** 5.0.0 (Docker image: ghcr.io/robotastic/trunk-recorder:latest)
- **Platform:** Raspberry Pi (docker-compose setup)
- **Blacklist file path:** `/data/talkgroup_blacklist.txt`
- **Config file:** `/app/config.json`

### What I Expect

No recordings for TGIDs listed in `/data/talkgroup_blacklist.txt`.

### What Happens

Recordings for blacklisted TGIDs (e.g., CATS Bus Ops 1, TGID 207) still appear in my scanner site and recordings folder.

### Evidence

#### Config Snippet

```json
"blacklistTalkgroupsFile": "/data/talkgroup_blacklist.txt",
```

#### Blacklist File Contents

```text
207
209
211
2137
2139
735
737
739
741
731
2149
1689
1691
1693
```

#### Sample Talkgroups.csv lines

```csv
Decimal,Hex,Alpha Tag,Mode,Description,Tag,Category
207,0cf,CATS Bus Ops 1,D,Bus Operations 1,Transportation,Charlotte Area Transit System
209,0d1,CATS Bus Ops 2,D,Bus Operations 2,Transportation,Charlotte Area Transit System
...
```

#### Docker Logs

No mention of blacklist loading or skipping:

```
docker logs trunkrecorder | grep blacklist
(no output)
```

### Steps Taken

- Confirmed config and blacklist paths.
- Ensured blacklist TGIDs match those being recorded.
- Updated to latest Docker image.
- Restarted TrunkRecorder.
- Still see recordings for blacklisted TGIDs.

### What should happen?

TrunkRecorder should skip/ignore calls for any TGID in the blacklist file.

---

**Please advise if this is a bug or if there is an additional config step required.**

taclane, 2025-07-26:
Trunk Recorder does not presently have a `blacklistTalkgroupsFile` option.  If there is an error in the configuration documents, let us know and we can make sure it is corrected.

The way this feature is currently managed is [through the talkgroup csv](https://github.com/TrunkRecorder/trunk-recorder/blob/master/docs/CONFIGURE.md#talkgroupsfile).  Open it up with excel or an equivalent editor, and add a "Priority" column to the top. Give everything a priority `1` that you would like to record, and a priority `-1` for talkgroups to ignore (it is important that each line has some value once you add the header).  

Export it back out as a csv, and Trunk Recorder will use this to record or ignore transmissions.

MeckMyDay, 2025-07-27:
Thank you so much!!

MeckMyDay, 2025-07-27:
It actually still doesn't work.I think I've done everything right. :(

***@***.***:~/trunkrecorder $ docker images | grep trunk-recorder
robotastic/trunk-recorder latest be3f76e38d12 3 months ago 647MB
***@***.***:~/trunkrecorder/data $ ls CharlotteUASI
encode-local-sys-0.sh talkgroups.csv   { "ver": 2,
"digitalRecordersPerSource": 4, "sources": [ { "center": 853812500, "rate":
2400000, "error": 0, "gain": 39, "digitalRecorders": 4, "driver":
"osmosdr", "device": "0" }, { "center": 857000000, "rate": 2400000,
"error": 0, "gain": 39, "digitalRecorders": 4, "driver": "osmosdr",
"device": "1" }, { "center": 852912500, "rate": 2400000, "error": 0,
"gain": 39, "digitalRecorders": 4, "driver": "osmosdr", "device": "2" }, {
"center": 854500000, "rate": 2400000, "error": 0, "gain": 39,
"digitalRecorders": 4, "driver": "osmosdr", "device": "4" }, { "center":
853000000, "rate": 2400000, "error": 0, "gain": 39, "digitalRecorders": 4,
"driver": "osmosdr", "device": "3" } ], "systems": [ { "control_channels":
[ 852912500, 853262500, 853350000, 853812500 ], "type": "p25",
"modulation": "qpsk", "shortName": "CharlotteUASI", "talkgroupFile":
"/data/talkgroups.csv", "useTalkgroupPriorities": true,
"recordUnknownTalkgroups": false, "uploadScript":
"/data/encode-local-sys-0.sh" } ], "captureDir": "/data" }

***@***.***:~/trunkrecorder $ cat docker-compose.yml version: '3'
services: trunkrecorder: image: robotastic/trunk-recorder:latest
container_name: trunkrecorder volumes: - ./config.json:/app/config.json -
./data:/data devices: - /dev/bus/usb:/dev/bus/usb ports: - "3000:3000" # ←
Add this line restart: unless-stopped

docker logs trunkrecorder | grep priority [2025-07-27 02:06:06.759756]
(info) ------------------------------------- Trunk Recorder
------------------------------------- [2025-07-27 02:06:06.759787] (info)
------------------------------------- INSTANCE
------------------------------------- [2025-07-27 02:06:06.759793] (info)
Using Config file: /app/config.json [2025-07-27 02:06:06.759797] (info)
Trunk-Recorder: Version: 5.0.0 [2025-07-27 02:06:06.759801] (info) Log to
File: false [2025-07-27 02:06:06.759809] (info) Log Directory: logs
[2025-07-27 02:06:06.759826] (info) Temporary Transmission Directory:
/dev/shm [2025-07-27 02:06:06.759832] (info) Capture Directory: /data
[2025-07-27 02:06:06.759836] (info) Upload Server: [2025-07-27
02:06:06.759841] (info) Broadcastify Calls Server: [2025-07-27
02:06:06.759845] (info) Status Server: [2025-07-27 02:06:06.759850] (info)
Instance Key: [2025-07-27 02:06:06.759854] (info) Instance Id: [2025-07-27
02:06:06.759859] (info) Broadcast Signals: false [2025-07-27
02:06:06.759866] (info) Default Mode: digital [2025-07-27 02:06:06.759870]
(info) Call Timeout (seconds): 3 [2025-07-27 02:06:06.759890] (info)
Control channel warning rate: 10 [2025-07-27 02:06:06.759897] (info)
Control channel retune limit: 0 [2025-07-27 02:06:06.759902] (info) Phase 1
Software Vocoder: false [2025-07-27 02:06:06.759908] (info) Enable Audio
Streaming: false [2025-07-27 02:06:06.759912] (info) Record Unit to Unit
Voice Calls: true [2025-07-27 02:06:06.759916] (info) New Call from UPDATE
Messages: true [2025-07-27 02:06:06.759922] (info) Frequency format: mhz
[2025-07-27 02:06:06.759927] (info) Status as String: true [2025-07-27
02:06:06.759933] (info) Log Level: info [2025-07-27 02:06:06.759939] (info)
Color Console/Logfile Output: console [2025-07-27 02:06:06.759946] (info)
------------------------------------- SYSTEMS
------------------------------------- [2025-07-27 02:06:06.759960] (info)
System Number: 1 ------------------------------------- [2025-07-27
02:06:06.759967] (info) Short Name: CharlotteUASI [2025-07-27
02:06:06.759972] (info) System Type: p25 [2025-07-27 02:06:06.759978]
(info) Control Channels: [2025-07-27 02:06:06.759984] (info) 852.912500 MHz
[2025-07-27 02:06:06.760000] (info) 853.262500 MHz [2025-07-27
02:06:06.760008] (info) 853.350000 MHz [2025-07-27 02:06:06.760014] (info)
853.812500 MHz [2025-07-27 02:06:06.760038] (info) Loading Talkgroups...
[2025-07-27 02:06:06.760043] (info) Talkgroups File: [2025-07-27
02:06:06.760050] (info) Modulation: qpsk [2025-07-27 02:06:06.760055]
(info) Conversation Mode: true [2025-07-27 02:06:06.760061] (info) Analog
Recorder Maximum Deviation: 4000 [2025-07-27 02:06:06.760066] (info) Filter
Width: 1 [2025-07-27 02:06:06.760071] (info) Squelch: -160 [2025-07-27
02:06:06.760075] (info) De-emphasis Tau: 0.00075 [2025-07-27
02:06:06.760080] (info) API Key: [2025-07-27 02:06:06.760084] (info)
[... 96 lines cut ...]
Talkgroups: false [2025-07-27 02:06:06.760165] (info) Minimum Call Duration
(in seconds): 0 [2025-07-27 02:06:06.760172] (info) Maximum Call Duration
(in seconds): 0 [2025-07-27 02:06:06.760177] (info) Minimum Transmission
Duration (in seconds): 0 [2025-07-27 02:06:06.760183] (info) Multiple Site
System: false [2025-07-27 02:06:06.760187] (info) Multiple Site System
Name: [2025-07-27 02:06:06.760192] (info) Multiple Site System Number: 0
[2025-07-27 02:06:06.760200] (info) [2025-07-27 02:06:06.760206] (info)
------------------------------------- SOURCES
------------------------------------- [2025-07-27 02:06:06.760215] (info)
Driver: osmosdr [2025-07-27 02:06:06.760219] (info) Device: 0 [2025-07-27
02:06:06.760227] (info) Center: 853.812500 MHz [2025-07-27 02:06:06.760237]
(info) Rate: 2400000 [2025-07-27 02:06:06.760245] (info) Error: 0
[2025-07-27 02:06:06.760251] (info) PPM Error: 0 [2025-07-27
[... cut ...]
Gain: 0 [2025-07-27 02:06:06.760281] (info) TIA Gain: 0 [2025-07-27
02:06:06.760286] (info) MIX Gain: 0 [2025-07-27 02:06:06.760290] (info) AMP
Gain: 0 [2025-07-27 02:06:06.760294] (info) VGA Gain: 0 [2025-07-27
02:06:06.760298] (info) VGA1 Gain: 0 [2025-07-27 02:06:06.760302] (info)
VGA2 Gain: 0 [2025-07-27 02:06:06.760306] (info) Idle Silence: 0
[2025-07-27 02:06:06.760384] (info) signal_detector_cvf_impl: samp_rate:
2.4e+06 fft_len: 1024 window_type: 0 threshold: -45 sensitivity: 0.9
auto_threshold: true average: 0.8 quantization: 0.01 min_bw: 0 filename:

MeckMyDay, 2025-07-27:
> There are a couple things to investigate within your config.
>
>    - It is supposed to be talkgroupsFile not talkgroupFile. Because of
>    that, it isn't able to load the .csv file you prepared
>    - When defining a rtl source, it should look like
>
>                         "driver": "osmosdr",
>                         "device": "rtl=1234,
>
> [where 1234 is the serial number or device index (0 through number of SDR
> devices -1)]
>
>    - "digitalRecordersPerSource": 4 is not a valid config option
>    - "useTalkgroupPriorities": true is not a valid config option

Thank you kindly!  Copilot/ChatGPT are no substitute for real knowledge!
Everything is working great now! Much appreciated!

closed 2025-07-27 by robotastic