TrunkRecorder/trunk-recorder #1032: Crashing. Possibly due to poor signal, but not sure.

ancker010, 2025-07-31:
Getting crashes randomly. Here's a paste of the logs.

```
[2025-07-31 22:19:47.019410] (error)   [starcom21peso] Retuning to Control Channel: 852.412500 MHz
[2025-07-31 22:19:47.019668] (info)   	 - System Source 0 - Min Freq: 850.957000 MHz Max Freq: 853.257000 MHz
[2025-07-31 22:19:47.034783] (info)   	 Xlating Channelizer decimator - freq_xlating taps: 91 Decim: 4 Resampled Rate: 24000 Lowpass Taps: 83
rtlsdr_read_async returned with -2
rtlsdr_read_async returned with -2
terminate reached from thread id: 76facd6006c0Got unknown exception
terminate reached from thread id: 76f8834006c0Got unknown exception
0x76fbc268293d:0x76fbc268293d: (gr::tagged_stream_block::check_topology(int, int)+0x142d)
0x76fbc20640da: (gr::tagged_stream_block::check_topology(int, int)+0x142d)
0x76fbc20640da: (std::rethrow_exception(std::__exception_ptr::exception_ptr)+0x8a)
0x76fbc204e8e6: (std::rethrow_exception(std::__exception_ptr::exception_ptr)+0x8a)
0x76fbc204e8e6: (__cxa_call_terminate+0x33)
0x76fbc20638ba: (__cxa_call_terminate+0x33)
0x76fbc20638ba: (__gxx_personality_v0+0x31a)
0x76fbc1eb2b06: (__gxx_personality_v0+0x31a)
0x76fbc1eb2b06: (_Unwind_GetTextRelBase+0x2c16)
0x76fbc1eb35cd:[2025-07-31 22:19:47.130680] (error)   [starcom21peso]	 Control Channel Message Decode Rate: 1/sec, count:  4
 (_Unwind_Resume+0x12d)
0x76fbc25c4a5c: (_Unwind_GetTextRelBase+0x2c16)
0x76fbc1eb35cd: -- error: unable to obtain symbol name for this frame
0x76fbc2c8d78d: (_Unwind_Resume+0x12d)
0x76fbc25c4a5c: (osmosdr::device::find(osmosdr::device_t const&)+0x5add)
0x76fbc2647db2: -- error: unable to obtain symbol name for this frame
0x76fbc2c8d78d: (gr::block_executor::~block_executor()+0x242)
0x76fbc2620a4d: (osmosdr::device::find(osmosdr::device_t const&)+0x5add)
0x76fbc2647db2: (gr::local_sighandler::throw_signal(int)+0x5f13)
0x76fbc268793b: (gr::prefs::save()+0x64b)
0x76fbc26879e4: (gr::block_executor::~block_executor()+0x242)
0x76fbc2620a4d: (gr::prefs::save()+0x6f4)
0x76fbc25d299f: (boost::detail::make_ready_at_thread_exit(boost::shared_ptr<boost::detail::shared_state_base>)+0x36f)
0x76fbc1d1aaa4: (gr::local_sighandler::throw_signal(int)+0x5f13)
0x76fbc268793b: (pthread_condattr_setpshared+0x684)
0x76fbc1da7a34: (gr::prefs::save()+0x64b)
0x76fbc26879e4: (__clone+0x44)
```

I'm running the latest version of "TheGreatCodeholio" docker image with mqtt built in. IIRC, it should be running the latest version of TR.

Happy to provide whatever else you need to debug.

lilhoser, 2025-08-05:
I'm seeing this too -- https://github.com/TrunkRecorder/trunk-recorder/issues/1034

robotastic, 2025-08-14:
@ancker010 Can you try capturing a core dump: https://trunkrecorder.com/docs/DEBUG
That will help us see where things are going south. This looks like it could be a HW or driver issue.
```
rtlsdr_read_async returned with -2
rtlsdr_read_async returned with -2
```
Generally means that TR is not able to talk with the SDR dongle.