TrunkRecorder/trunk-recorder #1035: MotoTRBO Capacity Max and Connect Plus Decoding for TR

s3m1s0n1c, 2025-08-11:
Hey Brains Trust...

Had a quick search thru issues and couldn't find anything regarding this so thought id start the convo of implementing it..

Could we pretty please have full trunking recording of Cap Max and Con Plus systems...

Thanking You :)
Sonic

SimonSheehan, 2025-08-12:
I'd very much like to see DMR improved to include slots or talkgroups as well, would be phenomenal.

robotastic, 2025-08-14:
This has been on the backlog of things I have been meaning to do. I do have a DMR radio... but no repeater nearby.  Does anyone has a recommendation for one of those local bridges you can use and plug into the internet?

robotastic, 2025-09-10:
> s3m1s0n1c
>  left a comment
> (TrunkRecorder/trunk-recorder#1035)
> Any update @robotastic <https://github.com/robotastic>?
>
> Did ya want me to provide baseband I/Q data for the MotoTRBO Capacity Max system that I listen too?
>
> Any MMDVM Hat is good.. Simplex or Duplex... I run a Duplex one with multiple HAM DMR Networks..

Step one is probably for me to just get support for DMR Repeaters and basic stuff like that working. Does this one look good enough?

https://www.amazon.com/AURSINC-Duplex-Hotspot-Support-Raspberry/dp/B089K195T2/ref=sr_1_8?dib=eyJ2IjoiMSJ9.x0QG8PTNFWGNO9ddkFEkZkSuXStrA0PU10lv3lrVVx-BbRWKVeUilqQFMjOAjhHUtWfvU7sT8HIKijITuLaJcr1_7KMvux6ejQnhsCONHIPan-a-0-gl1mvWGQHBhPKk66XraU1KRnhwxuyv5kUEFbJsFil-vJttrQSzdLl7DGhLRScYnl5ZgKO99b0s-It3ea2aPp8DmSjN9T9vX4fh9Dq3nVsboPT22FvcgPiqZJQ.-T8g-TTwSuA6pzKphWFuDWqYNY84U7zwXsX-yiAOOhI&dib_tag=se&keywords=mmdvm&qid=1757503571&sr=8-8&th=1￼
AURSINC MMDVM Duplex Hotspot Module Dual Hat V1.5.2 with 0.96 OLED Display Support P25 DMR YSF NXDN DMR Slot 1 + Slot 2 for R Pi 1, 2/2b, 3/3b/3b+, 4/4b, Pi Zero, Zero 2W with USB Port (with OLED)
amazon.com